Ultimaker/Cura #8524: cura 4.7.1

Artfil3D, 2020-10-12:
1er bug: a chaque fois que je découpe une pièce creuse un tube ou un pot, cura considère la pièce comme pleine et prévoit un remplissage a l'intérieur, la fonction support est désactiver. Donc désinstallation de cura, réinstaller le logiciel est cela persiste, donc je suis passé a la version 4.5.0 et le problème est résolu.
2eme bug la fonction support manuel sur cura 4.7.1 me rajouté une fois la pièce découpé des support tout autour de celle-ci, impossible de construire les support manuellement et cela sous différente pièces ou figurines.

nallath, 2020-10-12:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

Ghostkeeper, 2020-10-13:
It would help to post your questions in English, and to create a separate issue for each bug.

You might want to check if "Remove All Holes" is enabled for you.
Note that Manual Support is a plug-in someone built for Cura. It's not something we maintain.

rburema, 2020-10-13:
_google-translated_

En plus de ce que @nallath a écrit ci-dessus:

Je suis désolé, mais aucun des développeurs ne parle assez bien le français, et la version google-translate de vos mots n'avait que très peu de sens.

Cependant, cela dit;
- Veuillez remplir correctement le modèle de bogue, en anglais si vous le pouvez.
- Veuillez joindre les modèles qui vous posent des problèmes (vous devrez peut-être les compresser pour que github les accepte).
- Nous ne corrigeons pas les bogues sur les anciennes versions (sauf s'il s'agit d'un risque de sécurité, et même alors seulement sur certaines versions), donc si nous avons déjà corrigé quelque chose dans une version plus récente, nous n'allons rien faire pour l'ancienne.

closed 2020-10-26 by no-response[bot]