Ultimaker/Cura #8525: Change Raft Layer Flow

Timocop, 2020-10-12:
**Is your feature request related to a problem? Please describe.**
Currently we are unable to change the (bottom) raft layers flow rate. "Initial Flow Rate" only affects the model first layer not the rafts first layer. Sometimes its better to dump more plastic to the build plate to increase adhesion such as ABS.

**Describe the solution you'd like**
Able to change rafts (first layers) flow rate.

**Describe alternatives you've considered**
Currently ChangeAtZ (Post Processing Script) is the only way to do this. However, post processing scripts do not work as good with raft enabled (due to negative layers in g-code). Also you are unable to save it into a config.

**Affected users and/or printers**
Anyone that uses heavily warping filaments such as ABS on a glass plate.

**Additional context**
-

Ghostkeeper, 2020-10-13:
Well you can, but the UX is a bit bad because the settings are organised differently for the 3 types of raft layers. Instead of adjusting the flow, you can directly adjust the line width for the raft. There is a separate setting for the line spacing, so increasing the line width effectively functions as if you're increasing the flow for other types of structures.

Normally (e.g. with skin) you get two settings that determine the position and volume for the lines:
* Increasing the Line Width causes each line to be printed to extrude more material (with the same line length and layer height, more material = wider lines). It also causes the lines to be spaced further apart.
* Increasing the Flow only does the first part of that: extruding more material, without spacing the lines further apart.

The raft layers have different settings though: Line Width and Line Spacing. Since Line Width doesn't affect the spacing, it's effectively just a setting for the flow.

The UX is a bit bad though, because people expect the line spacing to change when changing the line width, and then to have a separate Flow setting to adjust the line width without affecting the spacing. So I think that is the feature we'd have to build.

Ghostkeeper, 2020-10-13:
For now, a workaround is to increase the Line Width, then set the Line Spacing back to what it was before adjusting the Line Width.

GregValiant, 2024-11-20:
A lot of changes were made to the raft settings in current versions.
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

closed 2024-12-08 by github-actions[bot]