Ultimaker/Cura #8526: [4.7.1] Some Post Processing Scripts behave differently to layer numbers

Timocop, 2020-10-12:
**Application version**
Cura 4.7.1

**Platform**
OS: Windows 10 Pro
CPU: Ryzen 5 3600
GPU: Nvidia RTX 2060 Super

**Printer**
Custom FFF Printer (Geeetech i3 Prusa Pro W)

**Reproduction steps**
1. Load a Model.
2. Enable Raft.
3. Add "Filament Change" post processing script and enter a layer number.
4. Look at exported g-code. Layers do not match the entered value.

*OR*

1. Load a Model.
2. Enable Raft.
3. Add "ChangeAtZ" post processing script and add layer number.
4. Look at exported g-code and see layer numbers do not match the cura preview layers.

**Screenshot(s)**
Filament Change script with Raft enabled:
[image]

**Actual results**
Every post processing script behaves differently to layer numbers. Very inconsistent and annoying to deal with.

**Expected results**
When entering a layer number it should be using the Cura preview layer numbers instead to avoid confusion.

**Project file**
-

**Log file**
-

**Additional information**
-

daT4v1s, 2020-10-13:
I didn't report this (yet) tho also had that problem
(temperature change, didn't notice when edge transition in layers not intended)

perhaps some combination of 
- UI layer view preview should add option to also indicate the g-code's layer (units)
- or allow change@Z script apply to UI coordinates instead

apparently if the offset was calculated, negative numbers may be used to apply
custom layer settings to supports and rafts

( apparently supports also count to the pre-base layer offset)

AbeFM, 2020-10-20:
There used to be issues with the Z-height being off by 1, you'd have to generate g-code and just check where the change was and offset the number till it put it in the right place.. But they fixed it. It looks like it came back when they combined change-at-z across all printer flavors.

I've created another issue with Change At Z that hasn't been a problem in years: #8574

Ghostkeeper, 2020-10-22:
Change At Z wasn't combined across printer flavours. Pause At Height was.

closed 2024-12-08 by github-actions[bot]