Ultimaker/Cura #8527: Custom part cooling fan curve

katherinehackworth, 2020-10-13:
**Is your feature request related to a problem? Please describe.**
Fan speed can only be set to a given percentage for the part cooling fan.

**Describe the solution you'd like**
The ability to set a custom fan curve, defining part cooling fan speeds based on the overhang angle of the part at that current area as so: [image] This is similar to how PC users are able to set custom fan curves for their PC case fans as a function of the (usually) CPU temperature.

**Describe alternatives you've considered**
Unsure.

**Affected users and/or printers**
All users with part cooling fans on their printers.

**Additional context**
N/A

Ghostkeeper, 2020-10-14:
From a user experience point of view, I think this would require some sort of tiered UI that allows people to use a simple version first without needing to understand these graphs. Otherwise this setting would not be understood by many users, and some users might bulk at the complexity.

mrx23dot, 2020-12-18:
+1 for overhang based fan control, I need 100% fan for overhangs, but 0% fan for the rest of layers with ABS.

halfshavedyaks, 2021-09-05:
I also want overhang fan control, but it needn't be a custom curve. As mrx23dot has said, one setting for any overhangs above a certain threshold would be adequate in many cases, analogous to the overhang speed setting. A default smooth gradient might be better still and a custom curve would be icing, but a basic version would do most of what I need.

Ghostkeeper, 2021-09-14:
Something similar is being discussed in https://github.com/Ultimaker/Cura/issues/5140, Halfshavedyaks.

GregValiant, 2023-07-02:
See Pull request #15853