Ultimaker/Cura #8531: [4.7.1] crash while loading project 3mf file

pilere, 2020-10-13:
**Application version**
(The version of the application this issue occurs with.)
4.7.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 last version

**Printer**
(Which printer was selected in Cura?)
CREALITY CR 10 S PRO

**Reproduction steps**
1. open 3mf file (previouly saved by cura but maybee previous version)
2. select open as project
3. validate "resumé"/substract of the project

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
Crash

**Expected results**
(What should happen after the above steps have been followed.)
loading of the file

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
`2020-10-13 18:30:45,040 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-10-13 18:30:45,080 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-10-13 18:30:45,107 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-10-13 18:30:45,131 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-10-13 18:30:45,156 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-10-13 18:30:45,181 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-10-13 18:30:45,232 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-10-13 18:30:45,261 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-10-13 18:30:45,294 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-10-13 18:30:45,314 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-10-13 18:30:45,335 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2020-10-13 18:30:45,365 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2020-10-13 18:30:45,385 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2020-10-13 18:30:45,406 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2020-10-13 18:30:45,430 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
2020-10-13 18:30:45,456 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PerObjectSettingsTool
2020-10-13 18:30:45,477 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
2020-10-13 18:30:45,500 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-10-13 18:30:45,521 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PreviewStage
2020-10-13 18:30:45,547 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RemovableDriveOutputDevice
2020-10-13 18:30:45,569 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-10-13 18:30:45,613 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimulationView
2020-10-13 18:30:45,639 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SliceInfoPlugin
2020-10-13 18:30:45,664 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-10-13 18:30:45,693 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-10-13 18:30:45,769 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-13 18:30:45,777 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-13 18:30:45,781 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-13 18:30:45,789 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-13 18:30:45,793 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-13 18:30:45,797 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-13 18:30:45,803 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-13 18:30:45,807 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-13 18:30:45,827 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin Toolbox
2020-10-13 18:30:45,851 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CameraTool
2020-10-13 18:30:45,889 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MirrorTool
2020-10-13 18:30:45,923 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RotateTool
[... 432 lines cut ...]
2020-10-13 18:49:43,496 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\default.shader]...
2020-10-13 18:49:43,510 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\selection.shader]...
2020-10-13 18:49:43,524 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\select_face.shader]...
2020-10-13 18:49:43,540 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\default.shader]...
2020-10-13 18:49:43,554 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\composite.shader]...
2020-10-13 18:49:43,571 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\overhang.shader]...
2020-10-13 18:49:43,587 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\striped.shader]...
2020-10-13 18:49:43,604 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\transparent_object.shader]...
2020-10-13 18:49:43,618 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\striped.shader]...
2020-10-13 18:49:43,633 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xray.shader]...
2020-10-13 18:49:43,646 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xray_composite.shader]...
2020-10-13 18:49:43,662 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\default.shader]...
2020-10-13 18:49:43,675 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\grid.shader]...
2020-10-13 18:49:43,692 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\platform.shader]...
2020-10-13 18:49:43,733 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xray.shader]...
2020-10-13 18:49:45,438 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 12.2s
2020-10-13 18:49:54,148 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-10-13 18:50:37,865 - INFO - [Thread-4] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2020-10-13 18:53:54,672 - WARNING - [MainThread] 3MFReader.ThreeMFWorkspaceReader.preRead [358]: Ignoring instance container [creality_cr10spro_0.6] with type [variant]
2020-10-13 18:54:04,489 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-10-13 18:55:16,558 - WARNING - [Thread-5] 3MFReader.ThreeMFWorkspaceReader.preRead [358]: Ignoring instance container [creality_cr10spro_0.6] with type [variant]
`

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-10-14:
Does this happen with all project files or just the one? How does it crash? Does it show a crash screen? It just exits onto desktop?

pilere, 2020-10-14:
only with this file, I have redone the project from scratch and it's loading fine, the file is probably corrupted ...

nallath, 2020-10-14:
Could you then also share the project file that causes it?

closed 2020-10-28 by no-response[bot]