Ultimaker/Cura #8535: Stuck on set up

walk5, 2020-10-14:
**Application version**
Version 4.4.1
[image]

**Platform**
Kubuntu 20.04.1

**Printer**
Robo R1+

**Reproduction steps**
1. In the terminal I entered "sudo add-apt-repository ppa:thopiekar/cura"
2. After the repository is added. I typed "sudo apt-get update"
3. After that step, I used "sudo apt-get install cura"
4. After installed, I tried to open the program but it is stuck on the install screen. 

**Actual Results**
The image shows where it is stuck. I installed cura before on another computer the same way. It was an older version. I am not sure which one. It worked great for some reason this time it is not working. It says to follow the steps but there is nothing to click on. I hope this is the tight place to ask this question. I tried to follow the outline provided. Please help. Thank you.

nallath, 2020-10-14:
We don't support the PPA, only the appimage. If you want support with the PPA, please contact the maintainer of the PPA.

closed 2020-10-14 by nallath

walk5, 2020-10-14:
Ok. Thank you for your help.

walk5, 2020-10-14:
@thopiekar Can you please help with this issue?

Ghostkeeper, 2020-10-15:
There is a separate issues page for it here: https://github.com/thopiekar/Cura-packaging/issues

The issue is most likely using the wrong Qt version with Cura.