Ultimaker/Cura #8538: "Print All Walls" option

Asterchades, 2020-10-14:
**Is your feature request related to a problem? Please describe.**
Recently I printed out a small base for a statuette. The base was made from a projected 2-bit image and featured a number of small gaps and small walls as a result. When it finally came to printing, the only options available were to have the narrow walls removed, apply a global horizontal expansion which closed up some of the narrow gaps, or try the Print Thin Walls option which can be a little hit-and-miss - more miss when it's exposed detail like this.

**Describe the solution you'd like**
I'd be keen to see the implementation of a "Print All Walls" option which took these overly thin walls and treated them as if they were of the minimum allowed extrusion size. This naturally is not dimensionally accurate but that is not the purpose of the feature, rather it is simply to ensure that all provided detail is present even if not specifically accurate.

As an extension, this option could also allow wall width multipliers - or at least a 1/2 wall option. Single extrusions aren't necessarily going to work on all printers, especially across multiple layers, so two walls may be required to actually make it worth doing in some cases.

**Describe alternatives you've considered**
The only real alternative I can think of, at least for now, would be a rework of the aforementioned "Print Thin Walls" option such that it tried its best to adaptively print the walls using less extrusion, which may already be on the card with Arachne. However as these will be exposed in the final product this is likely to produce low quality results in the case of particularly thin walls, such as anything below about 80% of the nozzle size.

**Affected users and/or printers**
I see this mostly affecting people who print objects with detailed patterns on top or bottom surfaces - primarily things such as key tags, badges, or perhaps some jewellery, outside of niche uses such as my own case. It's probably not a huge crowd so I wouldn't expect a lot of resources to be devoted to it.

**Additional context**
[Warrior Crest.zip](https://github.com/Ultimaker/Cura/files/5378738/Warrior.Crest.zip)
This is the exact model I was attempting to print. Sliced by default, with a 0.4mm line width, this will lose a number of the smaller raised lines in the final layers, while applying a 0.2mm horizontal expansion (which was how I printed it) closed up (though not completely) a number of the narrow gaps such as in the wrapping in the handle at the top.

With my idea, at least as it is in my head, this could be printed such that the removed detail is present but the narrow gaps are completely unaffected (obviously some of the larger gaps would be slightly reduced in places).

rburema, 2020-10-15:
> or try the Print Thin Walls option which can be a little hit-and-miss - more miss when it's exposed detail like this.

We're working on (also, amongst many other things) improving this in this giant PR: Ultimaker/CuraEngine#1210. It'll make it possible to vary wall width (within an extrusion path), and thus be a _lot_ less hit-and-miss with the 'Print Thin Walls' option.

Asterchades, 2020-10-15:
Arachne is one I'm very much looking forward to, but I still think it's going to have a practical lower limit when it comes to exposed features. I don't see extruding a 0.1mm line reliably with a 0.4mm nozzle, for example, though I can't say I've actually tried.

Ghostkeeper, 2020-10-15:
CuraEngine#1210 also actually contains this precise feature request. See the sub-pull-request here that introduced the setting: https://github.com/Ultimaker/Cura/pull/8189 and specifically the setting there "Enable Variable Line Widening".

rburema, 2020-10-15:
@Asterchades Where the limits sould be is something we're still working out.

Asterchades, 2020-10-15:
> It will widen the wall then to be printable with the minimum line width, sacrificing dimensional accuracy.

Really? Interesting. I wasn't aware that it was an intended part of the feature, but that would definitely satisfy the request.

closed 2021-12-14 by Ghostkeeper