Ultimaker/Cura #8544: Keep active visible Settings while loading a project

Waldkaeuzchen, 2020-10-15:
I dont know if this is a bug or new feature, so I am posting this as a feature request

**Is your feature request related to a problem? Please describe.**
I have multiple settings which I use for my printer, which are normally not Visible in a standard Profile (like Draft Shield). When I open a project from a colleague or customer, which doesnt have the same settings visible, my visibility settings are overwritten and I have to add all of them again. 

**Describe the solution you'd like**
Maybe there could be a dialog asking if I want to add the visible settings from the loaded project to my settings or vice versa.

**Describe alternatives you've considered**
As an alternative the visible settings from the current Profile could be merged with the ones from the loaded project

**Affected users and/or printers**
Users who work with projects from other persons

**Additional context**
(Add any other context or screenshots about the feature request here.)

konskarm, 2020-10-15:
Would the [Setting Visibility Creator](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/SettingVisibilityProfileCreator) plugin do the job that you need?

Waldkaeuzchen, 2020-10-15:
Yes, thank you for the link

closed 2020-10-15 by Waldkaeuzchen