Ultimaker/Cura #8545: Problem between two printers

Fabdu57, 2020-10-15:
Hello,

I have a problem...
Two profiles, the first created with Ender 3 pro, the second with SOVOl SV02.
Same parameters. Same values. However, support are different, and i have some problem on certain parts...

[image]

Is somebody can help me, because if it continue, i will sell SV02 to take again a creality ender 3 lol.

Thanks in advance,
Fabrice

[Fichier 3mf.zip](https://github.com/Ultimaker/Cura/files/5385635/Fichier.3mf.zip)

eskeyaar, 2020-10-16:
These doesnt have project files, so pls upload them as project files.

Fabdu57, 2020-10-16:
on zip file you have the 3mf file.
You don't arrive to upload it ?

eskeyaar, 2020-10-16:
yes, those are just 3mf file, it doesnt open as project file.
Save them as project not export them.

Fabdu57, 2020-10-16:
OK sorry.
I will do it again tomorrow

eskeyaar, 2020-10-16:
also mention what version of cura you are using.

Fabdu57, 2020-10-17:
Version is 4.7.1
New files in attachment
[Files.zip](https://github.com/Ultimaker/Cura/files/5395229/Files.zip)

Ghostkeeper, 2020-10-19:
> and i have some problem on certain parts...

What is the problem you're having? Is it not filling that gap? Could you fill in the issue template, specifically with what your expected behaviour is and what your actual behaviour is?

I'm getting the same result with both project files, visually:
[image]

Inspecting the g-code there are still slight differences. You say that all of the settings are the same, but there still are differences. Most importantly, the Maximum Resolution and Deviation settings are still very different which has an influence on the thicknesses of all parts and the exact coordinates of your vertices. Try setting those to be equal.

closed 2020-11-02 by no-response[bot]