Ultimaker/Cura #8548: CURA does not change the language of the description dialog!

lehuuloc-0104, 2020-10-16:
[image]
Hello,
I changed the language on Cura 4.7, everything is working fine. But when I hover my mouse over the parameter boxes, the description appears in English, not in the language I chose.
Can you give me a fix?

Ghostkeeper, 2020-10-19:
Indeed, I could reproduce this.
The descriptions are translated; Cura has a translation. It's just not showing the translation for some reason.

I'll show this bug to the team to see what they think.

lehuuloc-0104, 2020-10-27:
> Indeed, I could reproduce this.
> The descriptions are translated; Cura has a translation. It's just not showing the translation for some reason.
> 
> I'll show this bug to the team to see what they think.

Thank you,
I look forward to this, hope it will be updated in the next Cura version.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]