Ultimaker/Cura #8555: Tuning - post processing at object change - once at a time

pflegezentrum, 2020-10-17:
Hi,

While tweaking settings like, line width or flow, one has to sweep throug several paramters. It would be quite handy if one could change seetings by script after an object changes in "on at a time" mode. 

Is this feature in, or could it be added. I gues this would be appriciated from a lot of makers out there!

BR
Gary Vedukovic

Ghostkeeper, 2020-10-19:
I'm not entirely sure what you mean. Sorry if language is a barrier.

You're finding that when using one-at-a-time mode, you need a different line width or a different flow rate?
What do the post-processing script have to do with that? Which post-processing scripts are you talking about?

pflegezentrum, 2020-10-20:
[image]

Changing all of them in the list ... ;) would be nice.

Ghostkeeper, 2020-10-22:
You mean a checkbox that enables all of the checkboxes?

pflegezentrum, 2020-10-27:
Hi,

Within CURA you can ste the print sequences to "n at a time" (See screenshot). 

[image]

It prints seperate opbjects in a sequence. so object 1.. n (See drawing)
[image]

For each Object 1 ..n wich is printed, it would be handy to change parameters. Like flow rate, temperature, line width. 
Thats the feature request. (See image with notes)
[image]

BR
Gary

Ghostkeeper, 2020-10-28:
Right, I think I understand. For each of these objects you'd want to change certain settings at a certain height using the ChangeAtZ script.

Ghostkeeper, 2020-11-02:
We've discussed this and decided to defer this issue, meaning that we realise it's a problem that could be improved but it doesn't get priority for now.

Reasons that factored into this discussion are:
* ChangeAtZ is only a tiny fraction of users' prints, and ChangeAtZ + one-at-a-time moreso.
* We fear that the interface to steer this would become very complex, with being able to select for which objects to make these changes.
* The Post-Processing Scripts system is currently not designed to track settings for each object separately, add sets of settings when you add objects, or delete them when the objects are removed from the build plate. So this change would have a significant amount of work.
* There is a workaround when you set the trigger to be at a layer number. The layer numbers keep counting on with multiple objects.

pflegezentrum, 2020-11-02:
Hi, 

Ok, and thanks for the info!

Please note:  
- ChangeAtZ is only a tiny fraction of users' prints, and ChangeAtZ + one-at-a-time moreso.?
A new option "ChangeAtObject" would be a new selection -> so not ChangeAtZ