Ultimaker/Cura #8556: [4.7.1] Nozzle & bed temp 5C higher in gcode than set in UI

jmattsson, 2020-10-17:
**Application version**
4.7.1

**Platform**
Windows 10, 64 bit

**Printer**
Anycubic Predator

**Reproduction steps**
1. Set bed and nozzle temp
1. Slice
1. Check resulting gcode

**Screenshot(s)**
[image]

**Actual results**
Incorrect temperature in gcode. The resulting temperature is 5C higher than it should be. Possibly related to the old #4537?

**Expected results**
Correct temperature in gcode.

**Project file**
n/a

**Log file**
No relevant information appears to be logged in the log file.

**Additional information**
n/a

Ghostkeeper, 2020-10-19:
That's the Initial Layer Build Plate Temperature and Initial Layer Print Temperature. They will change to your desired temperatures for the second layer.

The initial layer temperatures are often slightly higher to improve adhesion with the build plate.

closed 2020-10-19 by Ghostkeeper

jmattsson, 2020-10-19:
Right. That explains it, but is also non-obvious. Thanks for the clarification - I'll make sure I expose those settings too then so I get the control I want.

Ghostkeeper, 2020-10-22:
Yeah, often there is a contradiction between making things work well out of the box/automatically, and making things work in a simple way. This is a pretty good example of that contradiction.

jmattsson, 2020-10-23:
For some reason it hadn't even occurred to me that there might be first-layer settings for temperature as well. Not sure why my brain didn't explore that possibility, considering all the other first-layer settings :(