Ultimaker/Cura #8557: ( 4.7.1) Pause at height isn't working with Ultimaker S3 5.8.1

Fukasx, 2020-10-17:
**Application version**
Cura 4.7.1

**Platform**
Mac OS 10.15.7

**Printer**
Ultimaker S3 5.8.1

**Reproduction steps**
1. Use the Post Processing Plugin for Pause at Layer height 

**Screenshot(s)**

[image]

**Actual results**
1. Error message from the printer
2. The print pauses at specific layer height but after resume it start at the wrong Z height.
3. Custom text message isn't displayed

[image]

[image]

**Expected results**
No Z gap, no Error message on the printer and display custom message on the Printer

**G-Code at Layer height 77**

`G0 X141.078 Y137.291
G0 X141.414 Y138.118
G0 X143.823 Y138.123
G0 X143.806 Y138.221
;TIME_ELAPSED:40637.343775
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 77
M117 Edelstahlkugel einsetzen.
M0 ; Do the actual pause
G92 E1056.8877
;LAYER:77
M204 S1000
M205 X10 Y10
;TYPE:WALL-INNER
;MESH:Bauteil1.stl(43)
G1 F1500 E1063.3877
G1 X143.037 Y138.154 E1063.39048
G1 X142.264 Y138.21 E1063.39327
G1 X141.495 Y138.394 E1063.39612
G1 X140.781 Y138.692 E1063.39891

Ghostkeeper, 2020-10-19:
I don't have an S3 at home, just an Ultimaker 3 which doesn't exhibit this issue.

I tried to reproduce the problem, but the g-code looks fine. It's similar to your snippet. The Z command is right before the last travel move of the layer. After that travel it pauses (upon the layer switch). Then it continues at the same Z coordinate. No skip of layers, all the g-code is still there.

I'll ask someone at work if they can test this on an S3. Perhaps the firmware does something weird with the active bed levelling.

Ghostkeeper, 2020-10-22:
My colleague tested this on his UMS3 at home and also didn't see this issue. There was a bit of a blob at the height of the pause but that sort of thing is hard to prevent when the nozzle gets a lot of time to ooze. There was no extreme underextrusion like what you're seeing.

One thing you could try on your side is to enable "Redo Layer" in the post-processing script. It's meant to get the flow started again after it has oozed a bit.

We did find a different issue while testing this. The printer firmware gives an error when you add a message to the pause. We've forwarded that to the team that develops the firmware for the S-line printers of Ultimaker.

closed 2020-11-05 by no-response[bot]

SquirrelEngineering, 2021-01-05:
I found the issue. In the "PauseAtHeight.py" Script, there is an additional 1mm added to current_z that is the problem. It is real easy to modify. The script is found in
 `Ultimaker Cura 4.x\plugins\PostProcessingPlugin\scripts`

I found all instances of "Z=" in "Move head back" and took out the additional 1mm added. 

` 

                    # Move the head back
                    if current_z < 15:
                        prepend_gcode += self.putValue(G = 1, Z = current_z + 1, F = 300) + "\n"
                    prepend_gcode += self.putValue(G = 1, X = x, Y = y, F = 9000) + "\n"
                    prepend_gcode += self.putValue(G = 1, Z = current_z, F = 300) + " ; move back down to resume height\n"
                    if retraction_amount != 0:
                        if firmware_retract: #Can't set the distance directly to what the user wants. We have to choose ourselves.
                            retraction_count = 1 if control_temperatures else 3 #Retract more if we don't control the temperature.
                            for i in range(retraction_count):
                                prepend_gcode += self.putValue(G = 11) + "\n"
                        else:
                            prepend_gcode += self.putValue(G = 1, E = retraction_amount, F = retraction_speed * 60) + "\n"

                    if current_extrusion_f != 0:
                        prepend_gcode += self.putValue(G = 1, F = current_extrusion_f) + " ; restore extrusion feedrate\n"
                    else:
                        Logger.log("w", "No previous feedrate found in gcode, feedrate for next layer(s) might be incorrect")

                    prepend_gcode += self.putValue(M = 82) + " ; switch back to absolute E values\n"

`

Here is the whole script I am using. 
`# Copyright (c) 2020 Ultimaker B.V.
# Cura is released under the terms of the LGPLv3 or higher.

from ..Script import Script

from UM.Application import Application #To get the current printer's settings.
from UM.Logger import Logger

from typing import List, Tuple

class PauseAtHeight(Script):
    def __init__(self) -> None:
        super().__init__()

    def getSettingDataString(self) -> str:
        return """{
            "name": "Pause at height",
            "key": "PauseAtHeight",
            "metadata": {},
            "version": 2,
            "settings":
            {
                "pause_at":
                {
                    "label": "Pause at",
                    "description": "Whether to pause at a certain height or at a certain layer.",
                    "type": "enum",
                    "options": {"height": "Height", "layer_no": "Layer Number"},
                    "default_value": "height"
                },
                "pause_height":
                {
                    "label": "Pause Height",
                    "description": "At what height should the pause occur?",
                    "unit": "mm",
                    "type": "float",
                    "default_value": 5.0,
                    "minimum_value": "0",
                    "minimum_value_warning": "0.27",
                    "enabled": "pause_at == 'height'"
[... 456 lines cut ...]
                        if firmware_retract: #Can't set the distance directly to what the user wants. We have to choose ourselves.
                            retraction_count = 1 if control_temperatures else 3 #Retract more if we don't control the temperature.
                            for i in range(retraction_count):
                                prepend_gcode += self.putValue(G = 11) + "\n"
                        else:
                            prepend_gcode += self.putValue(G = 1, E = retraction_amount, F = retraction_speed * 60) + "\n"

                    if current_extrusion_f != 0:
                        prepend_gcode += self.putValue(G = 1, F = current_extrusion_f) + " ; restore extrusion feedrate\n"
                    else:
                        Logger.log("w", "No previous feedrate found in gcode, feedrate for next layer(s) might be incorrect")

                    prepend_gcode += self.putValue(M = 82) + " ; switch back to absolute E values\n"

                # reset extrude value to pre pause value
                prepend_gcode += self.putValue(G = 92, E = current_e) + "\n"

                layer = prepend_gcode + layer

                # Override the data of this layer with the
                # modified data
                data[index] = layer
                return data
        return data
`