Ultimaker/Cura #8561: Secuence printing, pauseAtHeight only working on the first piece

gerardcuadras, 2020-10-18:
**Application version**
4.7.1

**Platform**
macOS 10.15.3 (Catalina)

**Printer**
Ender 3

**Reproduction steps**
So the idea is to print same piece multiple times. One each time, also I want to stop at some height on every piece to change filament. So the idea is for example, start with orange filament, stop the printer, change filament to white, finish the piece and start the next one in white. After that, pause printer, change filament, and same as before. 

I placed the pieces on the plate, check what heights I want to stop and add a row for each piece. (second screenshot, only two rows for test purpose)

**Screenshot(s)**
[image]
[image]

**Actual results**
Pauses at the first specified height, however on the second piece don't stops. Same for the rest of the pieces.
So I checked the gcode and it's only adding the first pause at height.

```
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 15
M83 ; switch to relative E values for any needed retraction
G1 F1500 E-4
G1 F300 Z4.2 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S210 ; standby temperature
M84 S0
M0 ; Do the actual pause
M109 S210 ; resume temperature
```

This is the only actual pause at height on the gcode, even though I added more. 

**Expected results**
Pause at height specified on every piece, not only the first one. 

**Additional information**
I noticed that the printer displays the layer of every piece,but when it's doing for example the third piece it shows "Layer 64 / 19" (19 is the actual layer height of a single piece).

Ghostkeeper, 2020-10-19:
The script is indeed unaware of the existence of one-at-a-time mode and just pauses at the first time it reaches that height.

Ghostkeeper, 2020-10-20:
Sorry, we've concluded that this issue can't get the priority it needs for us to solve this now. This gets put on the backburner, in case the priorities change. The usage of this combination of features is very small.

closed 2023-06-24 by github-actions[bot]