Ultimaker/Cura #8562: After installing 4.7.1, printers and profiles were deleted.

BarmerO, 2020-10-18:
After installing 4.7.1, printers and profiles were deleted. How to restore it all.

nallath, 2020-10-19:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

Ghostkeeper, 2020-10-19:
If you checked the option to "erase all existing configuration" in the uninstaller of the old version, then the configuration is erased and not recoverable.

BarmerO, 2020-10-19:
> If you checked the option to "erase all existing configuration" in the uninstaller of the old version, then the configuration is erased and not recoverable.

I did not check this box.

After launch, a message came out:

Configuration error
Your configuration data is likely corrupted. I'm having problems with the following profiles:
Kossel Mini fdmextruder_user

Do you want to reset to factory default settings? Resetting will erase all current printers and profiles.

Ghostkeeper, 2020-10-20:
I tried starting Cura 4.6, adding a Kossel Mini, then upgrading to 4.7. The bug didn't reproduce.

Could you fill out the issue template? It's important to be able to debug your issue.

closed 2020-11-03 by no-response[bot]