Ultimaker/Cura #8563: Weird behavior printing

jonwienke, 2020-10-18:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
I sliced the flow calibrator STL. When it gets to the layers with the 3 concentric walls, the middle and inner wall print normally. Then the nozzle retraces the middle wall as if printing it again, but does not advance the filament. After retracing the entire middle wall, the outer wall prints, but like crap, because the nozzle has been oozing while retracing the middle wall, so there is a gap in the outer wall until the filament has advanced enough to replenish what oozed away while doing the WTF.

I've tried pretty much every permutation of Z seam settings, made sure ironing is turned off, etc, and nothing changes the behavior.

**Screenshot(s)**
N/A

**Actual results**
The STL prints, but quality is crap. Lots of layer gaps in the outer wall.

**Expected results**
The nozzle should print each wall in sequence, and follow the path for each wall line once per layer, instead of multiple times.

**Project file**
[CuraBug2020-10-18.zip](https://github.com/Ultimaker/Cura/files/5397807/CuraBug2020-10-18.zip)
Includes STL, project file, and defective sliced gcode.

**Log file**
N/A

**Additional information**
N/A

nallath, 2020-10-19:
Does it also happen if you don't change any settings? 

I'm also not quite sure what you exactly mean. Could you make a screenshot with some annotations to clarify what you mean?

jonwienke, 2020-10-19:
It happens no matter what settings I use.

A screenshot doensn't really show the problem. Open the sliced gcode file, and look at any layer 6 or higher on Octoprint or any other simulator. The printhead is doing a totally pointless and unwanted extra trip around the perimeter between printing the inner and outer walls, and the outer wall is getting screwed up as a result.

There are 3 walls in the upper layers, so the nozzle should trace the perimeter of the circle 3 times--once for each wall. But it's doing so 4 times, It prints the middle wall, prints the inner wall, does the unnecessary ring aroung the rosy, then prints the outer wall. And the oozing during the ring aroound the rosy borks the outer wall.

Ghostkeeper, 2020-10-20:
This is because the model has a curve that's almost exactly 3 line widths wide. Cura's walls are only closed contours, which always produces an even number of walls if you count both sides of the shape. So what's happening is that it actually produces 4 walls (2 outer, 2 inner), but one of the walls gets its flow reduced to almost 0 by Compensate Wall Overlaps.

The fix is to set the Minimum Wall Flow to anything higher than 0%. Setting it to a reasonable 50% eliminates the 4th wall and just makes it draw 3 circles.

On top of that, it seems that in two places along the contour the width dips below 3 line widths due to badly staggered vertices on the inner surface and outer surface, causing Cura to fit only 2 lines there. This is only for an almost invisible length and won't affect the print much if Minimum Wall Flow is used, but if not it causes the inner wall to break up into 2 closed loops.

closed 2020-10-20 by Ghostkeeper

jonwienke, 2020-10-20:
Minimum wall flow is set to 5% in this case, not 0%.

And setting minimum flow to 50% just makes the middle wall disappear completely. Only the outer and inner walls print, leaving a wall-width gap between them. So you may wish to look at what is happening again, and re-evaluate.

Ghostkeeper, 2020-10-22:
I will look again and re-evaluate.
Your project file sets Minimum Wall Flow to 0%:

[image]

It's not being overwritten by a profile or customised, so it's the default which is 0% for the Creality Ender 5 Plus.

Overriding it to 50% doesn't make the middle wall disappear completely, at least not as far as I can see. It does move all over the place courtesy of the seam alignment being set to "Random". And like I said before, the width dips below 3 line width in two places along the ring. It properly creates three walls, as I record here: http://dulek.net/work/3_walls.webm

So my evaluation is the same!

jonwienke, 2020-10-24:
Please review the attached gcode and project file, and show me where the middle wall went.

[Bug II.zip](https://github.com/Ultimaker/Cura/files/5433092/Bug.II.zip)

Ghostkeeper, 2020-10-27:
Indeed in this new project file there is no inner wall in the ring any more. This has nothing to do with Minimum Wall Flow though. Resetting Minimum Wall Flow to 0% doesn't make it come back. This is because your new project file makes the model thinner through Horizontal Expansion.

Your new project file sets the Horizontal Expansion to -0.036mm and the Hole Horizontal Expansion to -0.003mm. Your wall is actually 1.204mm wide. The negative horizontal expansion makes that 1.132mm. The Hole Horizontal Expansion on top makes that 1.135mm. Your line width is 0.4mm, so there can only fit 2 lines (0.8mm) in the 1.135mm width.
Resetting these two settings to 0mm makes the wall 1.204mm wide, which fits 3 lines (1.2mm) with 0.004mm to spare. So the fourth line gets a compensated flow rate of 0.004mm / 0.4mm = ~1% flow, although at that scale the rounding errors start getting quite significant so it'll probably vary by a few percent.

jonwienke, 2020-10-27:
If that's the issue, then it would make far more sense to print the middle wall at ~98% flow, rather than omitting it entirely. Particularly when minimum wall flow is set to less than 100%. The current behavior is a bug, not a feature.

A wall should only be omitted if it is narrower than the minimum wall flow setting. And if there are multiple walls side by side that don't work out to a precise multiple of line width, it would be preferable to print them 80-80-80-80% than 100-100-20-100%, to avoid omitting walls unless absolutely necessary.

Ghostkeeper, 2020-10-28:
I would call something a bug if it's not behaving as expected by the documentation (or the developers if that documentation is missing). That is something different from whether it behaves perfectly. If the perfect behaviour is the expected behaviour, then nothing can be considered a new feature. Everything is bugs. Not doing non-planar layers is a bug. Not having variable line width is a bug. That's not a good way to classify them, I'd say.
The current expected behaviour is that pieces smaller than a line width are not printed at all, unless Fill Small Gaps is enabled or (outside of the outer wall) if Print Thin Walls is enabled.

A new feature or enhancement changes the expected behaviour. Your preference of printing at 80-80-80-80% is going to be implemented by Ultimaker/CuraEngine#1210 if Variable Line Width Strategy is adjusted away from the default.
I think the optimal behaviour in this case would be to print at 100-60-60-100% flow, in your example. That is the default in Ultimaker/CuraEngine#1210.

jonwienke, 2020-10-28:
I had Print Thin Walls enabled IIRC.

100-60-60-100% makes sense, particularly if keeping the outermost wall thickness at 100% reduces surface artifacts and/or keeps dimensions more consistent.

To avoid gaps, I would also consider allowing going slightly wider than the nominal, e.g. 100-105-105-100% to avoid having tiny gaps when you have a 1.6mm wide model segment and a small positive Horizontal Expansion value.