Ultimaker/Cura #8565: Layers/chunks missing in support, and noisy supports

lukehutch, 2020-10-19:
**Application version**
4.7.1

**Platform**
Linux Fedora 32

**Printer**
AnyCubic i3 Mega

**Reproduction steps**
Download files from here:
https://www.thingiverse.com/thing:196038

Try slicing file "20150903 WingedVictory Full Scale Decimated to 1MTris.stl"

Use fine settings (0.1mm), and I think everything else on default triggers the problem

**Screenshot(s)**
[image]

**Actual results**
Note the gaps in the support column at the top right, and the weird noise sticking out the side of the model on the bottom right of the screenshot

**Expected results**
Support should not look like this

**Project file**
Linked above

**Log file**
Only the following lines appear in the log when slicing:

2020-10-18 23:14:44,066 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [710]: Slicing took 21.17879819869995 seconds
2020-10-18 23:14:44,066 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [711]: Number of models per buildplate: {0: 1}
2020-10-18 23:14:44,067 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [725]: See if there is more to slice...

lukehutch, 2020-10-19:
There's a similar problem with tree supports -- it creates some supports that just appear randomly in the middle of a space. The four layer "blob" in the middle of the following screenshot is several millimeters away from any other material around or below it.
[image]

konskarm, 2020-10-20:
Hi @lukehutch. I tried reproducing the original issue with the High 0.1 quality profile on an AnyCubic i3 and I do not observe it. Support seems to generate fine and it doesn't have any gaps. 

I would like to ask you for the following:
- Please provide a project file (File -> Save Project or Ctrl+S) that clearly displays the problem. This will make sure that it includes all the settings that you change, as well as your printer configuration
- When posting a picture of the Preview Mode, make sure to select Color Scheme -> Line Type, as it makes things clearer
[image]
- Your last comment that is related to a Tree Support should end up in a separate ticket, so that there is a different discussion for that (since it is probably a different bug). In this case, since there is already a ticket for supports in mid-air (#8570), please post your comment there to keep the discussions in the related issues

konskarm, 2020-10-20:
I stand corrected, I can see the problem on the front side of the model. I will put it up for discussion with the team.
[image]

lukehutch, 2020-10-20:
OK, great, I assume you have all the information you need then. Thanks for reproducing.

I'll assume that the problem with floating tree supports in my first comment is better covered by bug #8570, so you can ignore that here.

Ghostkeeper, 2020-10-20:
Looks like a duplicate of https://github.com/Ultimaker/Cura/issues/7191 to me. The Support X/Y distance is removing parts of support up until the point that it disappears. The support should actually be resting on the model, but having too high of a X/Y/Z distance setting makes the gap too big.