Ultimaker/Cura #8566: Tree supports sometimes don't touch the thing they're supposed to support

lukehutch, 2020-10-19:
**Application version**
4.7.1

**Platform**
Linux Fedora 32

**Printer**
AnyCubic i3 Mega

**Reproduction steps**
Download files from here:
https://www.thingiverse.com/thing:196038

Try slicing file "20150903 WingedVictory Full Scale Decimated to 1MTris.stl", with tree support enabled

Use fine settings (0.1mm), everything else default

**Screenshot(s)**
[image]

[image]

[image]

**Actual results**
Tree support on back does not touch the part of the dress that is flaring out behind the Winged Victory statue (three angles shown above so you can see it doesn't touch). 

**Expected results**
The support should either touch, or is useless.

**Project file**
Linked above

fieldOfView, 2020-10-19:
When inspecting supports, it is much easier to see what is going on if you change the color scheme from 'material color' to 'line type'.

Ghostkeeper, 2020-10-20:
I'm assuming you rotated the model to be upright, and scaled it so that it's not actually 3 metres tall, as with the original STL? What scale did you scale this to? A project file would still help to reproduce your issue.

lukehutch, 2020-10-20:
Yes, correct -- it's significantly shrunk, and rotated upright. You can find the project file linked in the latest comment in #8565. Just change the support type to tree, and keep all the other settings the same.

lukehutch, 2020-10-20:
Yes, correct -- it's significantly shrunk, and rotated upright.

Here's the last screenshot again, with line types colored: (this is at the back of the model)

[image]

Notice how the tree support contacts the surface at the bottom of the image, but does not contact the surface higher up.

There are other places where the supports do not meet the model, such as under the left wing:

[image]

Under the right armpit:

[image]

Right hip -- most of the support material is superfluous, and is doing nothing as it is too far away from the model:

[image]

Top of right leg -- support material suddenly appears out of nowhere (is not supported below) and has quite a big gap to the thing it appears to support

[image]

[image]

Sometimes the supports are not even underneath the things they're supposed to be supporting, e.g. top side of left leg:

[image]

etc. -- I'll let you examine the rest of the model for support issues.

I printed this anyway, and found that the supports failed frequently under the bottom edges of the wings, which left those edges "furry"-looking. I think that's because the gap between the supports and the model are often large -- and the size of the gap is very inconsistent.

Here's the project file:

https://gofile.io/d/1jNXd8

Ghostkeeper, 2020-10-20:
The problem seems to be significantly reduced if the Support Roof is disabled.

Ghostkeeper, 2020-10-22:
The existing issue with the tree support Z distance (CURA-7624, for Ultimaker developers) is currently in the 25th position of priority on our backlog. I've added this result to that as a related issue that should be fixed at the same time as #5759. I think the two issues are related and possibly fixed with the same fixes.

wjk, 2021-11-12:
@Ghostkeeper My apologies for the request, but has there been any update on fixing this bug? It still occurs in Cura 4.12.0, and your last comment indicated that you’d get to it in a few weeks… back in October of 2020. There has been no activity on this issue since. This problem is still preventing me from being able to use tree supports on my Prusa. I would greatly appreciate an update. Thanks so much!

Ghostkeeper, 2021-11-19:
Indeed, there has not been any progress with this. Other stuff has been put in between. It's currently still high up in the backlog, but some of the things that are higher are shrinkage compensation, alternating wall direction, the ongoing Marketplace redesign, support for MacOS Monterey and a few various print quality bugs found by our process engineers when tuning our profiles. Also notable that your recent bug report got placed higher than this one: https://github.com/Ultimaker/Cura/issues/10523

nallath, 2021-12-16:
It should be fixed for 4.13!

closed 2021-12-16 by nallath

wjk, 2021-12-18:
> It should be fixed for 4.13!

As of CuraEngine commit `86b36711517bb425403ebe660b42dac2e76bb6d2`, no it hasn’t been. There is still just as large a gap between the support and the model as before, as shown below. Please advise.

[image]

wjk, 2021-12-18:
@nallath I’ve built and installed the latest 4.13 CuraEngine, only to get the following wonderful message (same file and settings as before). What can I do to help diagnose this?

[image]

wjk, 2021-12-20:
I’ve built the 4.13 branch from source. I still see a sizable gap between the tree support and the object, as shown below. You said that this has been fixed in 4.13. What’s going on?

[image]

Ghostkeeper, 2021-12-29:
I don't know which version you compiled, but this is the result I'm getting with your project file from https://github.com/Ultimaker/Cura/issues/10523 on 4.13:
[image]

The project file specifies a Minimum Support X/Y Distance of 0.2mm, but due to the way sharp corners are visualised in the front-end, this gap gets blocked from view a bit. There is also a 0.15mm top distance specified, which is visible from this angle.

Here is the result with all distances set to 0 (looks very similar):
[image]

And with the defaults:
[image]

wjk, 2021-12-29:
@Ghostkeeper I thought I deleted that post. I had compiled the 4.13 engine, but forgot to run `make install`, so I was still running the old version. 😅 Sorry about that!