Ultimaker/Cura #8567: Outline rendering is sometimes broken

lukehutch, 2020-10-19:
**Application version**
4.7.1

**Platform**
Linux Fedora 32

**Printer**
AnyCubic i3 Mega

**Reproduction steps**
Download files from here:
https://www.thingiverse.com/thing:196038

Try slicing file "20150903 WingedVictory Full Scale Decimated to 1MTris.stl"

**Screenshot(s)**
[image]

**Actual results**
Blue outline doesn't extend all the way around the object. The area where the outline does not show changes as you rotate the view.

fieldOfView, 2020-10-19:
This is caused by the toolhandles. The toolhandles have an additional clickable volume around them. The area that is occupied by the toolhandles gets subtracted from the outline.

Ghostkeeper, 2020-10-20:
Yeah, it's been like this for a long time. Still a bit weird, though.

Ghostkeeper, 2020-10-22:
We've discussed this with the team. The reason this is happening is because the selection buffer is used to determine the outline (a blur on the selected object) and then the tool handles are also drawn in that selection buffer. Because of course, clicking those tool handles needs to make the tool handles react rather than selecting the object.
Fixing this would need us to change that to render two different buffers, one to determine the outline for the blur and another as actual selection buffer. We don't think it's worth investing the time to fix that problem then, and it could also incur performance issues.

We're going to defer this until we come across a better idea to fix it (no performance impact) or until priorities change.

closed 2023-06-24 by github-actions[bot]