Ultimaker/Cura #8568: Resizing is extremely touchy

lukehutch, 2020-10-19:
**Application version**
4.7.1

**Platform**
Linux Fedora 32

**Printer**
AnyCubic i3 Mega

**Reproduction steps**
Load an STL file into Cura. Switch to resize mode. Try dragging one of the axis arrows. Moving just one pixel changes the model scale by ~20%. It's very difficult to get it the size you want, just using the arrow.

nallath, 2020-10-19:
Do you also have a video of this? I don't have the issue (and also on fedora), so we need a bit more information to debug this.

lukehutch, 2020-10-19:
Sure, here's a video, but it's just exactly what I described.

[screen-recording.zip](https://github.com/Ultimaker/Cura/files/5405126/screen-recording.zip)

nallath, 2020-10-20:
Uh, that's really weird. Do you have any specific weird graphics setup? High DPI?

lukehutch, 2020-10-20:
Nvidia binary drivers with Radeon 1050 Ti, on Fedora 32, default DPI on 2560x1600 lodpi monitor (should be 90 dpi). Nothing weird at all.

fieldOfView, 2020-10-20:
> Nvidia binary drivers with Radeon 1050 Ti,

> Nothing weird at all.

I find that a pretty weird GPU. 
(I think you meant a Geforce 1050 Ti)

lukehutch, 2020-10-20:
Haha, yes, sorry, that was a 4am typo/thinko :-)  Definitely a Gefore 1050 Ti, not Radeon.

fieldOfView, 2020-10-20:
Is the model already scaled at the start of the screen-recording, or was it at 100% at the start?

Ghostkeeper, 2020-10-20:
It's already scaled (the original is almost 3m tall). And I think therein lies the problem here.

Ghostkeeper, 2020-10-20:
The problem reproduces also with this 1x1x1m cube: [1m_cube.zip](https://github.com/Ultimaker/Cura/files/5408213/1m_cube.zip)

fieldOfView, 2020-10-20:
In that case the workaround is to load the model, scale it down using the scale inputfields to something manageable (eg 10%), and then save the model out as STL and load it back into Cura. An alternative to saving and loading would be to use the "Apply transformations to mesh" function in the MeshTools plugin.

lukehutch, 2020-10-20:
Correct, the model was scaled down from 3m tall to ~15cm tall at the start of the recording. Looks like the scale changes are inversely proportional to the original size.

closed 2024-12-08 by github-actions[bot]