Ultimaker/Cura #8570: [4.7.1] Tree support starts in mid air

hrafnkelle, 2020-10-19:
**Application version**
4.7.1

**Platform**
Windows 10 64bit

**Printer**
Creality Ender3 Pro

**Reproduction steps**
1. Set up a fresh copy of Cura 4.7.1
2. Add a new printer using the Creality Ender 3 Pro profile and activate it
3. Enable "Support structure" visibility so "Tree" can be selected
4. Load the STL from https://cdn.thingiverse.com/assets/b8/71/d6/67/94/5015_Single_Radial_Fan_Dual_Ducts.stl
5. Set Support structure to Tree and Support Placement to "Touching buildplate"
6. Slice and observe the tree support generated in mid air in the opening for the fan

**Screenshot(s)**
[image]

**Actual results**
After slicing a support is generated that can not be printed

**Project file**
[HMG5_Tree_Slicing_Bug_Ender3.zip](https://github.com/Ultimaker/Cura/files/5403396/HMG5_Tree_Slicing_Bug_Ender3.zip)

lukehutch, 2020-10-19:
Looks related to https://github.com/Ultimaker/Cura/issues/8565#issuecomment-711616301

Ghostkeeper, 2020-10-21:
This issue occurred when the branches of support halfway down decided to rest on the model because they couldn't find a path to the build plate, even though higher up there was still a path available. If the Support Placement setting was set to Touching Buildplate, the branch disappeared from the moment where it decides it needs to rest on the model.

That issue is fixed in 4.8. The issue doesn't occur any more then.
[image]

The beta of 4.8 isn't far off any more, so you'll soon be able to check out the fixes and see if all of your issues with tree support have been eradicated.

closed 2020-10-21 by Ghostkeeper

jledgett, 2021-03-21:
I am running 4.8.0 and this tree starting in mid-air is still occurring.

See this screenshot, where a tree support was started in mid air at layer 36 where there is a clear view to the raft below.

[image]

I only realized this after the print was to that point and started purging a chunk of filament in that area on every pass.

maruhe, 2021-03-29:
same here, cura 4.8.0
4day print, running since a day, now I saw my Tronxy is printing in the air - may he will get it done
[image]

Ghostkeeper, 2021-04-13:
Please refer to https://github.com/Ultimaker/Cura/issues/4342 for those issues. This was a different issue where a branch suddenly stops in mid-air as it's projected downwards. Your issues are with the edge of the branch hanging in mid-air as the branches converge together on the bottom.