Ultimaker/Cura #8571: Print experiencing a y shift in same location across multiple prints. The shift occurs at weird gaps in the slicing when looking at the gcode file

greenarrow24, 2020-10-19:
**Application version**
(The version of the application this issue occurs with.)
Ultimaker Cura 4.7.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
This occurs on the newest windows (Windows 10)

**Printer**
(Which printer was selected in Cura?)
Tenlog D3 Pro
**Reproduction steps**
1. This error occurs with a change in quality
2. first we reset the printer to its original tenlog d3 pro cura profile
3. we noticed that when we raised the quality and sliced, we got those weird gaps i mentioned in the title and the print would shift about 3 mm in the y direction at these gaps.
4. we tried uninstalling and reinstalling the software to no avail
5. we tried tightening all pullies on printer to no avail.
6. we tried importing the cad as a different file type (originally as an obj) to no avail

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
(This shows the gap i was talling about in the slicing. )

[image]
(This shows what happens when we print at 0.15 quality)

**Actual results**
(What happens after the above steps have been followed.)
The pictures above show what happens when importing using an OBJ file and slicing at 0.15 quality with all other settings set to the original tenlog d3 pro cura profile.

**Expected results**
(What should happen after the above steps have been followed.)
What should happen is the part looks like this picture.
[image]

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
I attached these files in the link below

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
https://github.com/ajevans99/3d-printer-issues

**Additional information**
(Extra information relevant to the issue.)
It does not matter which way we have oriented the model in cura. We have tired on every side and every possible rotation but the issue still happens at the same place.

nallath, 2020-10-20:
Could you also share the model / project file?

ajevans99, 2020-10-20:
> Could you also share the model / project file?

You can find the obj, gcode, and log files here:
https://github.com/ajevans99/3d-printer-issues

(I'm working with @greenarrow24 on this)

Ghostkeeper, 2020-10-21:
Hmm, in your log file I find evidence that it's indeed finding a layer empty, which is probably due to being too small to print according to CuraEngine:

```
2020-10-18 17:00:42,168 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Warning: Couldn't find previous extruder plan so as to set the standby temperature. Inserting temp command in earliest available layer.
2020-10-18 17:00:42,172 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Empty extruder plan detected! Discarding extrusion temperature command.
```

However I'm not able to reproduce this problem myself. The model shows fine here:
[image]

> first we reset the printer to its original tenlog d3 pro cura profile

Cura has no profile for "Tenlog D3 Pro". So we're really just guessing in the dark as to what settings could be causing this. Could you provide us with a project file? That's under File -> Save... and will contain the exact settings that you sliced this with, and the exact models and their position/rotation/scale/etc. so it should reproduce your bug on our side.

closed 2020-11-04 by no-response[bot]