Ultimaker/Cura #8573: [4.7.1] Cura goes around arcs with widths that are odd multiples of the line width one extra time extruding very little filament

tjader0, 2020-10-19:
**Application version**
4.7.1

**Platform**
Windows

**Printer**
Ender 3 Pro

**Reproduction steps**
1. generate test stl file in OpenSCAD using attached code.
2. Slice it in Cura with 0.2 layer and 0.4 line width

**Actual results**
Cura goes around the circle one extra time when the wall is an odd multiple of the line width, one of them extruding very little filament.

**Expected results**
Cura goes around the circle exactly as many times as needed.

**Project file**
[test.zip](https://github.com/Ultimaker/Cura/files/5404032/test.zip) including scad, stl, gcode and 3mf

**Log file**
[test.log](https://github.com/Ultimaker/Cura/files/5404036/test.log)

**Additional information**
I am working on a model that is made mostly of straight and arc walls of constant width, so to get it to print faster I'm making the walls be a multiple of the line width I'm going to print with. However, when the walls are an odd multiples of the line width Cura goes around the arcs one more time than expected, while extruding very little filament. That is making the print take much longer than necessary and also causing some blobs to form.

To better isolate the problem I have made a simple OpenSCAD files that generates circles with walls that are from 1 to 5 times the line width, Cura prints the 2- and 4-walled circles as expected, but goes the extra time around in the 1-, 3- and 5-walled circles.

Ghostkeeper, 2020-10-22:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8563

Cura currently always creates closed shapes for its walls, and so if you were to take any scanline crossing them you'd always cross an even number of walls. The two inner walls overlap almost completely in your case of an odd multiple of the line width. Due to Wall Overlap Compensation, one of those inner walls gets reduced to almost 0% flow.
There is a setting that instructs Cura to omit walls below a certain flow rate: Minimum Wall Flow. Setting that to something like 40% or 50% will cause those low-flow-rate walls to be removed, which makes it actually print three loops in your ring.

There is a different routine in progress to fix this problem more elegantly: Ultimaker/CuraEngine#1210 . This is a big project though that will take us a few extra months before it can be released. But with the current release and a few releases to come, this is expected behaviour.

closed 2020-10-22 by Ghostkeeper

tjader0, 2020-10-22:
Thanks, I'll check the Minimum Wall Flow setting and look forward to the more elegant solution.