Ultimaker/Cura #8574: 'Change At Z' [5.2.1] does not respect "subsequent layers"

AbeFM, 2020-10-20:
**Application version**
CURA 4.7.1 Change At Z 5.2.1

**Platform**
win10

**Printer**
Custom dual extrusion and Custom IDEX

**Reproduction steps**
1) Add "change at z" post processing script
1a) set change to "target layer + subsequent layers"
2) slice object
3) Observe g-code

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
G-code contains the commands to set the requested "change at z" but then unsets them a few lines of g-code later:
I searched g-code for M104
`
H:\Shark230-190.gcode (13 hits)
	Line 17: M104 S230
	Line 58: M104 S230.0 T0
	Line 66: M104 S230.0 T0
	Line 665: M104 S225
	Line 14214: M104 S220.0 T0
	Line 14219: M104 S225.0 T0
	Line 29458: M104 S210.0 T0
	Line 29463: M104 S225.0 T0
	Line 46694: M104 S200.0 T0
	Line 47117: M104 S225.0 T0
	Line 77608: M104 T0 S0 ; turn off extruder
	Line 77609: M104 T1 S0 ; turn off extruder
	Line 77631: M104 S0`

**Expected results**
`H:\Shark230-190.gcode (13 hits)
	Line 17: M104 S230
	Line 58: M104 S230.0 T0
	Line 66: M104 S230.0 T0
	Line 14214: M104 S220.0 T0
	Line 29458: M104 S210.0 T0
	Line 46694: M104 S200.0 T0
	Line 77608: M104 T0 S0 ; turn off extruder
	Line 77609: M104 T1 S0 ; turn off extruder
	Line 77631: M104 S0`

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I can find other files like the project file (does it include post-processing?) upon request. I have tried this with both settings, it's not inverted, it always sets the temperature back across different printers, days, etc.

AbeFM, 2020-10-20:
How hard is that mistake to get to? It looks like an easy enough issue I might be able to handle it even with my meager coding skillz.

I'd be open to attempting developing a more programmatic parameter setting like start temp/temp step size/step mm size if it goes well.

Ghostkeeper, 2020-10-22:
I'm not able to reproduce this problem. Here is a project file that I created by executing your reproduction steps:[change_at_z_temperature.zip](https://github.com/Ultimaker/Cura/files/5421698/change_at_z_temperature.zip)

In the g-code I'm getting the following results (output from `grep -n M104` which also sets line numbers in front):

```
1054:M104 S200          ;<-- Normal printing temperature.
9771:M104 S199.0 T0     ;<-- Change At Z setting it to 199 degrees.
19038:M104 S200.0 T0    ;<-- At the end of the print, Change At Z seems sets it back. Unnecessary but harmless.
19044:M104 S0           ;<-- CuraEngine sets the temperature to 0 degrees at the end of the print.
19045:M104 T1 S0        ;<-- Also for other printers.
```

Do you have a project file that reproduces your problem?

The post-processing scripts are generally quite simple as far as programming goes. The Change At Z script in particular though has had many iterations and many features crammed into it by other people with meagre coding skillz so it's not the easiest one.

I'm not sure if I'd be open to having Change At Z perform multiple adjustments across multiple layers, in the interest of keeping the scope of the script contained, the code maintainable and the behaviour of the script clear. There are many things that people can do with Change At Z and a calibration gradient is just one of them. It's been rejected before.

AbeFM, 2020-10-23:
We should drop the additional functionality  as it confuses the conversation.

I'm getting different results today than when I first reported this. I have additional variables I need to sort out:

1) Running two copies of Cura at the same time (1 per printer as switching isn't seamless)

2) I installed more plug-ins since the previous test

Now I am not getting stuff turning the temperature back. But any temp change that's "this layer only" doesn't get done at all. I have 5 changes, and any ones I set to "target layer" just don't get one.

[PostProcessingProblems.3mf.txt](https://github.com/Ultimaker/Cura/files/5430943/PostProcessingProblems.3mf.txt)

I feel I'm not being a lot of help here. Can you suggest anything I can do to shed additional light? The way it's working now is useful, if not correct.

Ghostkeeper, 2020-10-27:
I don't think your two variables should really influence the test here, unless you change a printer in one instance of Cura that is active in the other. It is a bit arbitrary which one of the two actually gets saved to disk then, and due to lazy-loading this can also influence an already-running instance of Cura.

> Now I am not getting stuff turning the temperature back. But any temp change that's "this layer only" doesn't get done at all. I have 5 changes, and any ones I set to "target layer" just don't get one.

We have something similar going on that we just fixed last Friday: https://github.com/Ultimaker/Cura/issues/8075
But apparently that was only broken for the speed setting.
Your project file seems to crash libSavitar when it loads! We happen to have a similar crash in our current sprint. Hopefully I can be of more help if that is fixed. We're trying to get that fixed before the 4.8 stable.

AbeFM, 2020-10-27:
Every time I update CURA, if I start from 'scratch' or not, within a few days, I get all sorts of errors about improper extruders. So it's hard to judge when a problem I'm having is... legitimate. It would be *AWESOME* if there was a 'purge bad definitions' button, but instead, you can only wipe out everything which encourages you to just ignore the error. Pretty off topic, but I feel I do a worse job reporting bugs for it. I mentioned it a year or two back and there didn't seem to be any interest in it. It's not a complaint, just letting you know - yeah, there may be bugaboos.

This isn't a major issue for me, personally. When the next release hits, I'll try towers again and tell you if I see them behaving reliably - and of course I'm willing to test anything you specifically think it would be helpful to you for me to try.

Till then I'm just posting this to see if others have the issue or know how to fix it.

Thanks again everyone.

GregValiant, 2020-11-15:
[25x25Cube ChangeTempAtHeight.zip](https://github.com/Ultimaker/Cura/files/5543376/25x25Cube.ChangeTempAtHeight.zip)
I set it to change at 5mm and this resulted.
;[CAZD:
M104 S190.0 T0
;:CAZD]
G0 X123.7 Y123.7 Z5.1
;TYPE:WALL-INNER
;[CAZD:
M104 S205.0 T0
;:CAZD]
Set to change at a Layer number wasn't a problem.

lukaskurz, 2020-11-28:
I am experiencing similar issues while trying to print a temp tower. The temp changes made by ChangeAtZ get undone and the temp is always set to the "print temperature"/"initial print temperature", which is 200°C in my case.

```gcode
....
G1 X95.763 Y126.247 E273.80782
G1 X95.711 Y126.16 E273.81119
G1 F2700 E268.81119
;[CAZD:
M104 S225.0 T0; this is the change done by my ChangeAtZ
;:CAZD]
G1 F300 Z1.6
G0 F12000 X103.386 Y116.459 Z1.6
;TYPE:FILL
;[CAZD:
M104 S200.0 T0; this is what cura does on it's own
;:CAZD]
G1 F300 Z1.4
G1 F2700 E273.81119
....
```
After reading @GregValiant's comment, I was able to fix this, by disabling "Z hop when retracted".

without the z hop:
```
....gcode
G0 F12000 X102.16 Y112.06
G1 F2400 X102.209 Y112.108 E290.657
G0 F12000 X102.569 Y112.108
;MESH:NONMESH
;[CAZD:
M104 S225.0 T0
;:CAZD]
G0 F300 X102.569 Y112.108 Z1.6
G0 F12000 X102.239 Y116.271
....
```

I am fairly new to 3dprinting and all this, so I don't really know what's going on in the gcode, but from what i can see, the issue seems to be related to the z-hopping.

lukaskurz, 2020-11-28:
Could this be maybe related to #8715 ?

peta, 2021-01-09:
> Could this be maybe related to #8715 ?

Good catch mister! 😉 It definitely IS related to #8715 and the whole topic thus has a high criticality in my opinion.

[image]

AbeFM, 2021-01-10:
That is a good catch! I also saw it "sometimes" but never put it together.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Comango104, 2021-02-10:
Can confirm, Z-Hop is the culprit.

AbeFM, 2021-02-28:
It's certainly making it hard to get a representative temperature tower. I keep forgetting it's not working.

If it can't be fixed, can you put in a notice that it doesn't work with z-hop? It took me a while to figure out this problem today.

GregValiant, 2021-03-01:
Without Zhops in the file and making every possible change for the "current layer only" I get this:
;[CAZD:
M140 S90.0
M104 S100.0 T0
M104 S110.0 T1
M106 S76
M221 S80
M221 S83 T0
M221 S85 T1
M220 S60
;PRINTSPEED 70
;RETRACTFEEDRATE 2700.0
;RETRACTLENGTH 6.0
;:CAZD]
The recovery area at the end of that layer is this:
;[CAZD:
M104 S210.0 T0
M104 S210.0 T1
M220 S100
;RETRACTLENGTH -0.0
;:CAZD]

The retractfeedrate was OK
The print speed was OK
Both hot end temps were OK
The fan was adjusted but did not reset - Repair Needed
The 3 flow rates adjusted but were not reset - Repair Needed
The bed temp was adjusted but did not reset - Repair Needed
The retractlength just didn't work at all. - Repair Needed

Being the lazy guy that I am (and since I have no responsibility here), if it was me, I'd change the name of "ChangeAtZ" to "ChangeAtLayer" and I'd throw out the option of "By Height".  Then I'd take "PauseAtHeight" and rename it to "PauseAtLayer" and throw out the "By Height" option in that one for good measure.  In one fell swoop they both work with z-hops, adaptive layers, and spiralize.  That just leaves the 4 items above on the hit list.  When those are repaired then both plugins would be ready for the release of Arachne.

ryancasler, 2021-03-07:
I am seeing that if I use layer number selection and have it set to that layer only, fan speed is not reverted back to the original setting at the start of the next layer.  I originally thought this was because I had an cooling unchecked.  So, I checked cooling and set the fan speed to 0 and the problem still happened.  So, then I tried with an initial fan speed of 50 with a change a layer  X only to 100 and it still never reverted back to 50 after that layer.  No, I do not have Z-hop checked.  It appears that in terms of fan speed, it never respects the "this layer only" setting.
 
I have been using this setting to turn my fan on for the first upper layer with skin for prints with a very large flat top (think box shape) when printing with ABS and PETG.  Since I was printing with a fan speed of 0 and a nozzle of 0.8mm I was getting terrible pillowing.  You're lines are twice as far apart than with a 0.4mm nozzle because the lines are twice as thick.  As a result an infill of 20% doesn't make a whole lot of lines.  And my prints were terrible.  So, I started cooling with a moderate fan strength for PETG and a very low fan speed for ABS but only for the first layer that contained skin at the top of the print.   But my top layers have been pulling off with my ABS prints.  I think it might be because they are still being cooled and I assumed they weren't.  I didn't notice that the plugin wasn't turning the fan back off.

(On a side note, a plugin that would turn the fan on to a particular setting only for the first layer of skin after infill would be amazing!!  I did it by hand today and it wasn't too bad since the whole print only had 4 areas of skin.  So, I only had to insert four M106's and four M107's but turning the fan back off for inner and outer walls got me much better layer adhesion with ABS but still prevented the pillowing.)

This is resolved pretty simply by just inserting another change at Z-height for the next layer.  Putting a disclaimer in  with regards to fan speed would be helpful.  However, I think completely removing that option all together would be the easiest thing to do.  Make it explicit that the change is made at that Z-height or layer number and that's it and If you don't have some other plugin or setting that will make a change at a subsequent layer, it will use that setting for the rest of the print.  This would eliminate any possible unintended conflict with other plugins as well.  Plus, future dummies like me won't be confused.  It would be explicit that it's up to you to change it back when and if you want it to be. This shi.......stuff.....is so hard to understand sometimes.  We should make it as simple as possible whenever we can.  ;-)

Thanks for an amazing piece of software!  Being relatively new to 3d printing I have tried just about every slicer out there and I keep coming back to Cura because it is the only one that makes sense to me and that I don't fight with (as much).  So, thank you!

Ghostkeeper, 2021-03-08:
I reckon that implementing a system that allows post-processing scripts to show disclaimers to the user is more work than fixing the issue itself.

AbeFM, 2021-03-09:
The snarky reply is 'magic away the bug, then'... But what I meant was a
tool tip or add it to the text box where you add these scripts. Just plain
text. It was meant to be a quick way to keep people from thinking CURA is
just broken.

Do you think it would be impossible to add a "copy line" function? Often I
want to make a related change with one difference many times - eg
temperature tower.

Adding "change nozzle 1 temp to 200 at layer number 50", then copying the
line five times and finally hand editing those lines with new numbers would
be easier and less error prone. Should I make a separate issue? Similarly
with the suggestion that first skin layer could use fan - that's a good
suggestion that shouldn't be lost.

How does one find the post processing code? I shouldn't touch code any more
than I should help my dentist, but I'd still like to take a look.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

ryancasler, 2021-03-13:
> I reckon that implementing a system that allows post-processing scripts to show disclaimers to the user is more work than fixing the issue itself.

Well, since the plugin can display whatever text it wants to within the settings window, add a note would be rather simple.  Certainly better than broken functionality.

Ghostkeeper, 2021-03-16:
The post-processing scripts are loaded dynamically (so that you can put your own scripts in the resource folder) so there is an interface layer between them and the Cura code. To display a disclaimer, you need to adjust that protocol, create a data structure to store the disclaimers and then show the correct disclaimer in the interface. It's not at all hard, but then again, neither is the post-processing script itself. Truth is though that on Ultimaker's planning this sort of thing is not usually high up on the priority list because it's not innovating, not marketable and not understandable for managers. I'd still like to fix it, but it just doesn't get enough attention.

> Do you think it would be impossible to add a "copy line" function? Often I
want to make a related change with one difference many times - eg
temperature tower.

Yeah, should be possible. I think that's been asked here on Github before but I can't find the issue at the moment. So if you can't find it either you'd best make a new issue.

> How does one find the post processing code? I shouldn't touch code any more
than I should help my dentist, but I'd still like to take a look.

The post-processing scripts are here: https://github.com/Ultimaker/Cura/tree/master/plugins/PostProcessingPlugin/scripts

AbeFM, 2021-03-16:
Thanks and thanks!

I'll check if there's any descriptive text *at all* and suggest the best
place I can think up to mention the limitation.

Perhaps if z-hops were labeled in the g-code (perhaps in verbose mode?) It
would be easier to filter those lines out.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

novamxd, 2022-08-17:
If the issue is the z-hop unintentionally triggering the height marker detection, a possible solution would be to convert the height to a layer and then resolve everything to layers. That would be easy enough.

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

novamxd, 2024-11-21:
Excellent question! I've been a terrible maintainer. I'll take a look this weekend

novamxd, 2024-11-25:
OK! So a bit of an update. I've done some digging and can reproduce the issue with the fanspeed [related to this bug](https://github.com/Ultimaker/Cura/issues/8886). 

I also found a few other bugs related to detection that I'm ironing out right now, including with the z-hop. If I don't get a fix during this week I'll definitely continue working on it next weekend.

novamxd, 2024-12-01:
Another update! I believe I've solved issues #8574 and #8886. I've done some improvements around targeting with Z and, hopefully, improved the clarity of the code some. I also fixed an issue where on some prints that subtly change the values (ex bed temp) over a layer the post processor doesn't change them all to the desired value.

The only thing I need/want to do now is write some unit tests for it, though I need to see how I'd go about doing that.

novamxd, 2024-12-02:
@GregValiant So I'm stuck. I can't seem to get the tests to run under the `/plugins` directory and I have no idea where to begin. It's been quite a pain to get it to the point where I can at least run `python -m test` and I have had zero success in getting anything to run in PyCharm thanks to the fact that Uranium is not included.

What should I do here?

GregValiant, 2024-12-02:
ChangeATZ is a post processor (not a plugin) and must be in the "scripts" folder of your Configuration Folder.

I use Cura to debug.  There is a plugin in the MarketPlace called "Extensive support logging" or some such thing.  You can also import "Logger" and "Message" into the script.  If the script will actually run, then sometimes adding a note to data[0] will give you a clue as to what might be going on.  QML errors in the Settings section of your script will keep a script from loading and they can be hard to find.  Errors within the script ( = instead of == ) are caught and show up in the log when the script runs.

from UM.Message import Message
from UM.Logger import Logger
Sprinkled within the code log lines can tell you how far the code has gotten before it's hung up.  The Log file will also indicate the line within the code that may be a problem.
Logger.log("d", "The message goes here...")  ; "d" would be "DEBUG", "i" is "INFO" and "w" is "WARNING"

A message will pop up after saving:
Message(title = "Change at Z", text = "Is not compatible with 'One at a Time' print sequence.").show()
This will add a message to the first data section of the gcode.
data[0] += "; >>>>>>>>>>>> my variable = " + str(my_variable) + "\n"
QML errors in the settings can be tough to find.
I use other apps to check sections of code to see if they work, but mostly I use Cura because everything is there.
It can be a long boring process.

novamxd, 2024-12-04:
@GregValiant There seems to be some misunderstanding here.

> I can't seem to get the tests to run
> \- novamxd

> The only thing I need/want to do now is write some unit tests
> \- novamxd

So I'll ask again. **How can I write (and validate) unit tests for the plugin?**

I'm done with the functionality and I did indeed use the `UM.Logger` to assist with development, including emitting some changes to temp files.

I cannot, for the life of me, figure out how to **run the unit tests** in the `/plugins`.

If I just run `python -m test` in the root directory it seems to just run the tests in the `/tests` folder.

novamxd, 2024-12-04:
I also ran `python -m unittest` and it just falls on its face saying `ModuleNotFoundError: No module named 'UM'`

novamxd, 2024-12-14:
@HellAholic So I'm trying to figure out what the "normal" way to build and test Cura is. I have Cura checked out into a folder. I can even successfully run everything up to and including `conan install . --build=missing --update -g VirtualPythonEnv` from [here](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source).

But that's it. **Nothing else works**. I'm running `Windows 11` and `Python 3.10`. 

`build/generators/virtual_python_env.ps1` does not exist in my situation, though `build/generators/virtual_python_env.bat` does.

Even if that works I'm still presented with 

```
ImportError while loading conftest 'C:\Users\NoVaM\workspace\Cura\tests\conftest.py'.
tests\conftest.py:9: in <module>
    from UM.Application import Application
..\..\.conan2\p\urani7cca9f9a32795\p\site-packages\UM\Application.py:10: in <module>
    from UM.Controller import Controller
..\..\.conan2\p\urani7cca9f9a32795\p\site-packages\UM\Controller.py:3: in <module>
    from UM.Scene.Iterator.DepthFirstIterator import DepthFirstIterator
..\..\.conan2\p\urani7cca9f9a32795\p\site-packages\UM\Scene\Iterator\DepthFirstIterator.py:3: in <module>
    from UM.Scene.SceneNode import SceneNode
..\..\.conan2\p\urani7cca9f9a32795\p\site-packages\UM\Scene\SceneNode.py:17: in <module>
    from UM.Signal import Signal, signalemitter
..\..\.conan2\p\urani7cca9f9a32795\p\site-packages\UM\Signal.py:23: in <module>
    from UM import FlameProfiler
..\..\.conan2\p\urani7cca9f9a32795\p\site-packages\UM\FlameProfiler.py:12: in <module>
    from PyQt6.QtCore import pyqtSlot as PyQt6PyqtSlot
E   ImportError: DLL load failed while importing QtCore: The specified module could not be found.
```

If I try to run either `pytest` or `python cura_app.py`.

It shouldn't be this hard to build and test a code package.

novamxd, 2024-12-16:
OK. After some debugging it looks like this is an issue with how python is being included in `venv`. [This answer](https://stackoverflow.com/a/47143737) indirectly lead me to get past the issue above, though I ran into another.

I solved the above issue by removing the `python3.exe` included in the `build/generators/cura_venv/Scripts` folder and just using my local install of `Python 3.13` (which I upgraded to as part of my debugging).

Now I'm faced with

```
Traceback (most recent call last):
  File "C:\Users\NoVaM\workspace\Cura\build\generators\cura_venv\Lib\site-packages\numpy\core\__init__.py", line 24, in <module>
    from . import multiarray
  File "C:\Users\NoVaM\workspace\Cura\build\generators\cura_venv\Lib\site-packages\numpy\core\multiarray.py", line 10, in <module>
    from . import overrides
  File "C:\Users\NoVaM\workspace\Cura\build\generators\cura_venv\Lib\site-packages\numpy\core\overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
        add_docstring,  _get_implementing_args, _ArrayFunctionDispatcher)
ModuleNotFoundError: No module named 'numpy.core._multiarray_umath'
```

Which I'm going to do some more debugging to solve.

novamxd, 2024-12-16:
OK. The issue with numpy is an incompatibility between numpy==1.26.1 and Python 3.13. I've downgraded to Python 3.12.8 and am testing with that.