Ultimaker/Cura #8575: Post Processing Plugin | Pause at height | Repetier (@pause) 1mm to hight after continue

EGOiST1991, 2020-10-20:
Bug: 
The Post Processing Plugin "Pause at height" has a bug with the Repetier Method where the Z-height is 1mm to high after continue.

**Application version**
4.7.1

**Platform**
Windows 10 | Intel Coree i5-4960k | Geforce 1070 

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. Open a File
2. Extensions
3. Post Processing Scripts
4. Pause at Height
5. Layer Number (not heigth)
6. Pause at Layer (any will do)
7.Method: Repetier

**Screenshot(s)**
[](https://i.imgur.com/Z1q1mXJ.png)

**Actual results**
;TIME_ELAPSED:207.767029
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 3
M83 ; switch to relative E values for any needed retraction
G1 F300 Z1.65 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15.65
M84 E0
press @pause now and filament change ; Do the actual pause
@pause
G1 F300 Z1.65
G1 F9000 X0 Y0
G1 F2400 ; restore extrusion feedrate
M82
G92 E125.5039
G92 E125.5039
;LAYER:3
M106 S127.5
M204 S1800
M205 X8 Y8
;TYPE:WALL-INNER

**Expected results**
mycomments:
---- Here i am at 0.65mm
G1 F300 Z1.65 ; move up a millimeter to get out of the way
---- here the script from cura decides to go a mm up to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15.65
----- 14mm up again
M84 E0
press @pause now and filament change ; Do the actual pause
@pause
----- Now im in the Pause
G1 F300 Z1.65
**----- Here i am 1 mm to hight now!!!! Here should stay 0.65mmF**
G1 F9000 X0 Y0
----- Jetzt bist du wieder bei x0 y0 aber 1mm zu hoch. Hier fehlt die Bewegung auf 0.65mm!!!!
G1 F2400 ; restore extrusion feedrate
M82

**Project file**
Any Gcode/STL will do - all will have the same result :) Test it yourself

**Log file**
No need, obvious bug in the Script

**Additional information**
The Post Processing Plugin "Pause at height" has a bug with the Repetier Method where the Z-height is 1mm to high after continue.

EGOiST1991, 2020-10-20:
I just noticed that you guys use Pythonscripts.
I work with python for a living and it was a smallfix - so here is my contribution:

Where the Head moves 1mm up (thats correct):
`411:# Move the head away
412:prepend_gcode += self.putValue(G = 1, Z = current_z + 1, F = 300) + " ; move up a millimeter to get out of the way\n"`

The wrong line starts here:
`458:  prepend_gcode += self.putValue(G = 1, Z = current_z + 1, F = 300) + "\n"`
The correct line:
`458:  prepend_gcode += self.putValue(G = 1, Z = current_z, F = 300) + "\n"`

I changed it formyself now. But maybe you consider to update your script :)

konskarm, 2020-10-20:
Feel free to provide us with a PR so that we can review and test it! Also your contribution will be properly credited in our codebase.

EGOiST1991, 2020-10-20:
I dont know what PR stands for. What is it written out? I will provide it - if i know what it is 👍🏻 
If its a github thing, please explain - i just sign in for this post :)

Do you mean my python file? If yes, i attach it.
(Fyi)
The script is allready tested by myself 
[PauseAtHeight.zip](https://github.com/Ultimaker/Cura/files/5407758/PauseAtHeight.zip)

EGOiST1991, 2020-10-20:
I did a PR. Im new to github (started today) so ill hope i did everything right!
PR #8575

If there is something wrong with my first PR please let me know.

Rolf-Smit, 2021-08-05:
This issue seems to have been merged quite a while ago, but I still experience this issue on Ultimaker Cura 4.10.

Generated gcode:

```
G0 F300 X12.365 Y160.551 Z9.8
G0 F11250 X25.15 Y156.101
;TIME_ELAPSED:8298.876800
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current z: 10.08
;current height: 10.08
M83 ; switch to relative E values for any needed retraction
G1 F300 Z11.08 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S0 ; standby temperature
M0 ; Do the actual pause
M109 S200 ; resume temperature
G1 F300 Z11.08
G1 F9000 X25.15 Y156.101
G1 F300 Z10.08 ; move back down to resume height
G1 F2700 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E340.39059
;LAYER:34
;TYPE:WALL-INNER
;MESH:Body3 v2.stl
G1 F2250 X209.85 Y156.101 E348.99099
```

Notice how the Z height is changed just before the pause to: 9.8, then the pause script is reporting a height of 10.08 which is 1mm off.

If I use the script with layer height instead of mm the gcode is correct (notice the z height being 10.36 before the pause and the script moving the nossle up to 11.36 and back to 10.36):

```
G0 F300 X222.791 Y165.451 Z10.36
G0 F11250 X183.422 Y165.079
G0 X25.15 Y156.101
;TIME_ELAPSED:8642.415182
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 36
M83 ; switch to relative E values for any needed retraction
G1 F300 Z11.36 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S0 ; standby temperature
M0 ; Do the actual pause
M109 S200 ; resume temperature
G1 F300 Z11.36
G1 F9000 X25.15 Y156.101
G1 F300 Z10.36 ; move back down to resume height
G1 F2700 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E853.29208
;LAYER:36
;TYPE:WALL-INNER
;MESH:Body3 v2.stl
G1 F2700 E858.29208
```

Do I need to update anything else, or is this bug back again?

EGOiST1991, 2021-08-06:
Yes you are right. 
(I fixed the Layerheight issue) 
And the same Issue appears in Line 494. This is a result of laziness in coding.

Are you able to receive a file from me? You need to replace it with a file in the Curafolder and it will fix your problem (till the next Update) I will take some time till my fix is in the next Update. 

The Issue:
Wrong line 494:

`                        prepend_gcode += self.putValue(G = 1, Z = current_z + 1, F = 300) + "\n"`

Correct line: 494:

`                        prepend_gcode += self.putValue(G = 1, Z = current_z , F = 300) + "\n"`

EGOiST1991, 2021-08-06:
To change it: 
Ultimaker Cura 4.10.0\plugins\PostProcessingPlugin\scripts
and replace this file: PauseAtHeight.py with my attached file.
[PauseAtHeight.zip](https://github.com/Ultimaker/Cura/files/6943403/PauseAtHeight.zip)

nallath, 2021-08-16:
It's in 4.11. The beta is out in a few days.

closed 2021-08-16 by nallath

EGOiST1991, 2021-08-16:
Thanks. Glad i could help again.

nallath, 2021-08-16:
> Thanks. Glad i could help again.

No, thank you! :D

jellespijker, 2021-08-17:
While testing I noticed that the movement after the pause will first move the head down and then move to the position before the pause. This could result in noticeable artifacts or accidentally push thin towers from the bed.) See the code below.

```gcode
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current z: 5
;current height: 5.0
M83 ; switch to relative E values for any needed retraction
G1 F300 Z6 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z20.0
M84 E0
@pause now change filament and press ; Do the actual pause
G1 F300 Z5
G1 F9000 X30.759 Y30.759
G1 F1800 ; restore extrusion feedrate
```
I have switched the lines in the script such that it will first move to the correct X, Y and then move down
```python
prepend_gcode += self.putValue(G = 1, X = x, Y = y, F = 9000) + "\n"
prepend_gcode += self.putValue(G = 1, Z = current_z, F = 300) + "\n"
```

As shown in the code below
```gcode
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current z: 5
;current height: 5.0
M83 ; switch to relative E values for any needed retraction
G1 F300 Z6 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z20.0
M84 E0
@pause now change filament and press ; Do the actual pause
G1 F9000 X30.759 Y30.759
G1 F300 Z5
G1 F1800 ; restore extrusion feedrate
```

commit a1243421 (2021-08-16): Fix Pause at height issue

Fixes #8575

Thanks to @EGOiST1991 for suggesting the fix!