Ultimaker/Cura #8579: Raft: First layers miss first few lines and have to travel back to finish it

logiclrd, 2020-10-20:
**Application version**
4.7.1

**Platform**
Windows x64

**Printer**
Custom build

**Reproduction steps**
1. Add model to build plate.
2. Ensure that Build Plate Adhesion Type is set to Raft.
3. Slice the model.

The resulting G code draws a perimeter around the first layer, and then almost always, in my experience, starts drawing the raft's lines partway into the first layer, so that when it reaches the far side of the model it has to return to the start.

Obviously many models will _require_ travel to take place, because some places have multiple different lines at the same X value (assuming the first layer is angled perpendicular to X axis). Travel cannot in general be avoided. But, sometimes really unnecessary travel is generated, which is slow and results in the head dragging across all of the raft lines printed so far.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

The part highlighted in red requires the head to travel all the way back along the model to complete the printing of the first raft layer.

**Actual results**
In some cases extreme amounts of unnecessary travel when printing the raft, because the surface is split unnecessarily into two parts, each of which prints from the middle outward.

**Expected results**
In simple cases, these are not split and print consecutively. This takes less time and avoids travel where the print head may actually collide with each already-printed line of the raft (since the raft is compensating for bed imperfections and has an extremely high flow rate, the lines can end up higher than the nozzle).

Ghostkeeper, 2020-10-22:
In general this is very hard to solve, even when long travel moves are allowed. Getting the optimal order in which to print these lines is an extension of the [Traveling Salesman problem](https://en.wikipedia.org/wiki/Travelling_salesman_problem) for which there is no known polynomial-time algorithm (and an exponential-time algorithm would take too long to compute). So we have to use a heuristic. Currently the heuristic for skin is as follows:
* Start at the line closest to a corner perpendicular to the line direction. This is one of the "endpoints" as how a human would see it. There is no guarantee that there is always one good side to start from (like when printing a many-pointed star shape) but it's a good guess to at least prevent one long travel move.
* After printing every line, continue with the line that is closest to the current position.

It seems that for the raft, the first part is not implemented. It is implemented for the skin to prevent a long travel move that creates a scar on the surface. It was never considered a problem since the raft doesn't need to be pretty. For the sake of improving adhesion, we can consider implementing that.

logiclrd, 2020-10-22:
Thanks for the detailed response. Yes, I can see that finding a _truly_ optimal path is a trickier problem, _though_ for _most_ models (yes, there are pathological cases), the total number of different segments is such that an exhaustive search of different orders could be completed in microseconds. But, it isn't the order of separate sections that was concerning me, but the fact that it splits single contiguous segments unnecessarily -- and, in practice it appears that travel with early raft layers often causes minor collisions with the print head because the raft line is so thick it rises up higher than the bottom of the nozzle, so if you have to travel, then fine, but if it's _completely_ unnecessary, it feels like it actually has the potential to negatively impact or maybe even fail a print (though I haven't in practice observed this).

Ghostkeeper, 2020-10-23:
> Yes, I can see that finding a _truly_ optimal path is a trickier problem, _though_ for _most_ models (yes, there are pathological cases), the total number of different segments is such that an exhaustive search of different orders could be completed in microseconds.

Well, be careful with that. Your own model seems to have about 6 line segments per cm in the X direction from your screenshot, and is about 24cm long in the X direction, which would result in 144 lines, each of which can be printed in two directions. There are about [1.24*10^293](https://www.wolframalpha.com/input/?i=144%21+*+2%5E144) ways in which to print 144 lines. If every atom in the universe would calculate the travel lengths of 1 million of these options every microsecond, it would take about [6.5*10^193 years](https://www.wolframalpha.com/input/?i=%28144%21+*+2%5E144+%2F+%286*10%5E79%29+%2F+10%5E12%29+seconds) to slice a single layer. Those thought experiments are fun :P

But indeed, it appears to split a single contiguous segment here. Although of course in some cases you're going to need to have a split, like when printing something in the shape of an X. But at least one of these splits can be prevented, which is what we're doing with the skin like I explained before. We could apply that to the raft lines too, and I think that's what you're looking for.