Ultimaker/Cura #8586: No module named 'pynest2d'

smartes, 2020-10-21:
After updating Cura doesnt start.

Traceback (most recent call last):
  File "/usr/bin/cura.real", line 25, in <module>
    import pynest2d # @UnusedImport
ModuleNotFoundError: No module named 'pynest2d'

**Application version**
(The version of the application this issue occurs with.)

**Platform**
Ubuntu 20

nallath, 2020-10-21:
Check the running from source guides: 
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows#pynest2d
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-macOS#pynest2d
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu#pynest2d

closed 2020-10-21 by nallath

joonghochoe, 2020-10-22:
I also had a similar problem in the Windows 10 development environment. As a result, my case was due to the location of nlopt.dll called by pynest2d.pyd.

I used python.exe from C:\...\cura-build-environment-master\bin. When building cura-build-environment, nlopt.dll is originally created in the cura-build-environment\bin folder.

However, pynest2d.pyd could not find nlopt.dll, so I moved nlopt.dll to cura-build-environment-master\lib\site-packages folder and then executed python.exe cura_app.py normally.

Ghostkeeper, 2020-10-23:
For me it's linking nlopt statically, so it's inside pynest2d.pyd.

sea-clocktower, 2021-02-28:
> Check the running from source guides:
> https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows#pynest2d
> https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-macOS#pynest2d
> https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu#pynest2d

What if I've built from standard latest cura-build-environment image?
Today was trying to build from cura-build repo by running the shell script

scripts/python3.5/linux/build.sh

I did not provide the environment variable so docker image has been pulled from the default location:
CURA_BUILD_ENV_DOCKER_IMAGE="${CURA_BUILD_ENV_DOCKER_IMAGE:-ultimaker/cura-build-environment:centos-latest}"

After it was built I'm having the same issue while trying to run cura.
It seems that something has changed in the latest cura-build-environment images, because I remember I did the same thing a while ago and I could run the app.

Ghostkeeper, 2021-03-05:
Hmm, we use that image ourselves for our CI as well and there it can import `pynest2d` fine.