Ultimaker/Cura #8587: Entering invalid character in Material display name causes crash during autosave

csharpengineer, 2020-10-21:
Running version 4.7.1.  

Reproduction:
1.  Manage materials
2.  Duplicate an existing material ("Generic PLA")
3.  Select new material that was created
4.  In Display Name field, enter "Brandon's PLA"
5.  Tab to new field
6.  A second or two later, autosave will try to save the settings and Cura will crash.

Traceback (most recent call last):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\AutoSave.py", line 61, in _onTimeout
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 722, in saveSettings
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerRegistry.py", line 461, in saveDirtyContainers
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 604, in saveContainer
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 126, in setDirty
    if base_file is not None and base_file != self.getId() and not registry.isReadOnly(base_file):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 321, in isReadOnly
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\LocalContainerProvider\LocalContainerProvider.py", line 188, in isReadOnly
    file_path = self._id_to_path[container_id]  # If KeyError: We don't know this ID.
KeyError: 'generic_pla_175 #2'

Thread 0x0000098c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000abc (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001d18 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000e94 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004178 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000033b0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000a7c (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002120 (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004008 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002394 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000032b8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004228 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000045d0 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004fdc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004110 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004d7c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000054a4 (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 393 in _logInfoWidget
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.7-exe\build\inst\bin\cura_app.py", line 166 in exceptHook
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 844 in run
  File "X:\4.7-exe\build\inst\bin\cura_app.py", line 234 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

csharpengineer, 2020-10-21:
Nevermind, I may have a corrupted copy of Cura.  I removed some machine configurations that I didn't use in order to make it load faster and this is probably causing it to crash.  I can't create a new material at all now, even without invalid characters.

closed 2020-10-21 by csharpengineer

fieldOfView, 2020-10-21:
Protip: use the Startup Optimiser plugin to disable unused profiles instead of manually removing files.

csharpengineer, 2020-10-22:
> Protip: use the Startup Optimiser plugin to disable unused profiles instead of manually removing files.

Thank you so much!  I am new to Cura and didn't know this existed.  That helped greatly.

kuon, 2020-12-17:
I have a similar issue. I deleted all my cura preferences and I still get this error when I create a new material:

```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cura/AutoSave.py", line 61, in _onTimeout
    self._application.saveSettings()
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 722, in saveSettings
    ContainerRegistry.getInstance().saveDirtyContainers()
  File "/usr/lib/python3.9/site-packages/cura/Settings/CuraContainerRegistry.py", line 461, in saveDirtyContainers
    self.saveContainer(instance)
  File "/usr/lib/python3.9/site-packages/UM/Settings/ContainerRegistry.py", line 604, in saveContainer
    container.setDirty(False)
  File "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py", line 126, in setDirty
    if base_file is not None and base_file != self.getId() and not registry.isReadOnly(base_file):
  File "/usr/lib/python3.9/site-packages/UM/Settings/ContainerRegistry.py", line 321, in isReadOnly
    return provider.isReadOnly(container_id)
  File "/usr/lib/uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 188, in isReadOnly
    file_path = self._id_to_path[container_id]  # If KeyError: We don't know this ID.
KeyError: 'generic_pla #2'

```

Steps:

- brand new cura install (4.7.1) (all preferences in `~/.config/cura and `~/.local/share/cura` removed
- duplicate a generic PLA
- wait a few seconds
- crash

Ghostkeeper, 2020-12-22:
That's not reproducing for me, @kuon. I'd really like to get to the bottom of this because that KeyError is one of the most frequent crashes we're seeing reported through the automated crash report system (when people press the "send" button).

Which printer did you add before duplicating the material? I tried Custom FFF Printer and Ultimaker 3, but the problem reproduced for neither of them.

kuon, 2020-12-23:
I can reproduce it this way:

- delete `.config/cura`
- start cura (4.7.1)
- accept all default to wizard, create custom FFF printer
- when wizard is closed, hit ctrl+k
- create a new material and rename it
- crash

[image]

I do not have this issue with the appimage version, so it might be related to some python or QT version.

Ghostkeeper, 2020-12-24:
If you delete `.config/cura` you should probably also delete `.local/share/cura`.

You're using some sort of Debian package then, or something, rather than the AppImage? I'm not sure how up to date those are. It's generally not something we really support.

kuon, 2020-12-24:
I did delete the files in .local too. I am using archlinux aur package (cura-git), which is built from master.
-- 
Nicolas Goy