Ultimaker/Cura #8588: Perca dos perfis

fabriciosilveira-sjc, 2020-10-21:
O meu app Cura para Windows perdeu todos os perfis de configuração de impressão.
Mesmo eu tentando importar os perfis de backup o cura não aceita.

[image]

fabriciosilveira-sjc, 2020-10-21:
O perfil de material da empresa eSUN está com bug.
Quando ele seleciono o PLA desta impressa os perfirs de configuração do Cura desaparecem.

nallath, 2020-10-21:
Could you please report your problem / issue in English?

fabriciosilveira-sjc, 2020-10-23:
The material profile of the eSUN company has a bug.
When he selects the PLA from this form, the Cura configuration profiles disappear.

Ghostkeeper, 2020-10-26:
eSUN's "PLA Pro" is listed as being of the PLA+ material type. Most likely your printer has no profiles for PLA+. Sadly, not all combinations of printer & material type have any profiles. This is expected behaviour.

See also https://github.com/Ultimaker/Cura/issues/8229 .

closed 2020-10-26 by Ghostkeeper