Ultimaker/Cura #8589: Overlap modifier "infill mesh only" creates internal shells

MrEd88, 2020-10-21:
**Application version**
4.7.1 / Ultimaker_Cura-master.20201002014541.AppImage (nightly)

**Platform**
Linux Mint 20 - GTX 1050TI

**Printer**
Custom printer - (Printer is an Ender3 Modified to Direct Drive with BMG and e3dv6 clones.)

**Reproduction steps**
1. Load model and create a modifier
2. Set modifier to overlap with "infill mesh only" and change infill density.
3. Slice and check. 

**Screenshot(s)**

Cutting Mesh:
[image]

Infill Mesh Only:
[image]

**Actual results**

Where the modifier is overlapping there are shells before the new infill value is applied. However when changing to "cutting mesh", the infill is modified without creating new shells as expected, but it cuts the model in the external perimeters (I guess this is expected by using cutting mesh).

**Expected results**
No internal additional shells should be created when using "infill mesh only" when the idea is changing only the infill density.

**Project file**

[CuraProject.zip](https://github.com/Ultimaker/Cura/files/5418742/CuraProject.zip)

**Log file**
N/A

**Additional information**
seems like this [PR](https://github.com/Ultimaker/Cura/pull/6574) is related

nallath, 2020-10-22:
I found the issue; If you override the wall line count in the global settings, the wall thickness set in the per object settings isn't taken into account properly.

MrEd88, 2020-10-22:
Ohh I see, that was discussed in the mentioned PR but I did not understand. I am one of those who thinks it is easier to change the wall line count than wall thickness, I normally use different layer widths, so it is better to change the line count for me, now I will change the wall thickness instead. Thanks.

rburema, 2020-11-03:
@MrEd88 The number 7793 here doesn't refer to something publicly available, it's a reference to our internal system.

PR #8685 (2020-11-09): Also add wall_line_count when adding a infill mesh
CURA-7793

Fixes #8589
commit dc944ff5 (2020-11-09): Also add wall_line_count when adding a infill mesh

CURA-7793
Fixes #8589
closed 2020-11-09 by Ghostkeeper