Ultimaker/Cura #8591: Visual glitch: Random layer appears in the first layer instead of its proper layer

Deses, 2020-10-22:
**Application version**
4.7.1

**Platform**
Windows 10 x64, Ryzen 3700X & Radeon R9 390x

**Printer**
Ender 3

**Reproduction steps**
1. Place STL
2. Inspect layers

**Screenshot(s)**
[image]
[image]

**Actual results**
The layer 327 is displayed in the wrong place, but the exported gcode *SEEMS* fine.

**Expected results**
The layer should be in its place.

**Project file**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-10-22:
Duplicate of https://github.com/Ultimaker/Cura/issues/7412.

closed 2020-10-22 by Ghostkeeper

Deses, 2020-10-22:
> Duplicate of #7412.

I see! I didn't saw that one when I searched for a similar issue. Well, I'm glad you already have this one tracked. :D

Ghostkeeper, 2020-10-23:
It's a difficult issue that has been happening for years. So all we're really doing with it is tracking it. As far as we know it's random and pretty rare, so that makes it extremely hard to debug.