Ultimaker/Cura #8592: Layer shift in support but not in the model

GregValiant, 2020-10-22:
**Cura Version**
4.7beta

**Platform**
Win10 home 64 bit

**Printer**
Creality Ender 3Pro

**Reproduction steps**
1. Slice the model
2. Printed the model
3. The two attached images show what happened.
4. The zip file contains the 3mf project file.
The "LayerShift" file is a screen shot of Cura.

[image]
[image]

**Actual results**
Layer shift in the support tree only
[image]

**Expected results**
No shift.

**Project file**
[Cyl3 VStack3mf.zip](https://github.com/Ultimaker/Cura/files/5419964/Cyl3.VStack3mf.zip)

(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**

**Additional information**
The same shift occurred in the support after mirroring the model.  It looks like the model will print fine since the support goes up the center.

Ghostkeeper, 2020-10-23:
Yes, it's doing this because you've set the Tree Support Collision Resolution setting to 2mm. That means that it's going to do very inaccurate collision checking and that it's allowed to shift the branch by up to 2mm every layer. Indeed that's not going to print well. The setting is getting a warning colour to indicate that you've probably set it too high.

A more reasonable value would be 0.2mm or so. It's going to take longer to slice though as a result.

closed 2020-10-23 by Ghostkeeper

GregValiant, 2020-10-23:
Thank you sir.  I was having a problem in that the tree support wanted to
come up on the exterior of the model.  That was not acceptable as the
finish may have had strings running to the support.  I got random in my
adjustments trying to get the support back inside the model.

Thanks again,
Greg