Ultimaker/Cura #8594: Bug Cura Master scaling a rotated model

Gonzo3so, 2020-10-22:
**Application version**
Cura Master

**Platform**
Windows 10

**Printer**
Custom, Artillery Genius

**Reproduction steps**
1 insert model
2 rotate model
3 scale model uniform to 135 % 
4 untick uniform scaling
5 scale x axis to 150 % 

**Screenshot(s)**

**Actual results**
cura bugs and changes more than 1 axis, I believe due to "memory" of first position before rotating the object

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**

**Log file**

**Additional information**

konskarm, 2020-10-22:
I followed the reproduction steps and I am unable to reproduce this. Could you share a project file where the issue is observed? 

We recently saw a similar thing where in strange rotations the scaling _may_ influence not just the axis you are changing, but also other axes (e.g. if you are scaling X, it would change the scale percentage of Y _but only by a tiny bit, <1% or so_) which we attribute to rounding errors, since the mm field wasn't changing. That issue we decided to defer for now, since it isn't really affecting the model. Could it be the same issue?

nallath, 2020-10-22:
Did you also update Uranium? We quite recently made some changes to the scaling tool and that is in the Uranium repo.

Gonzo3so, 2020-10-22:
Hello, thanks for the quick reply.

It is my first bug report so if I didn't file it correctly I'm sorry.

I have included screenshots from the actions I took, hope this helps.

Friso

nallath, 2020-10-22:
Attaching things via email doesn't work; We can't see them. If you want to share pictures, you will have to do so via the github webpage.

Gonzo3so, 2020-10-22:
[image]
[image]
[image]
[image]
[image]

Included you find the screenshots, sorry complete noob here for filing a report. No I havent updated Uranium, I just downloaded the zip file but not sure what to do with it or how to use it.

konskarm, 2020-10-22:
Wait, you are saying you are using Cura master. Are you running Cura from source, or just running the latest cura version (4.7.1)?

Gonzo3so, 2020-10-22:
This is the cura I run 
[image]

I double checked and the version 4.6.1 I have also has the same bug.

Ghostkeeper, 2020-10-23:
That doesn't really tell which version it really is :(
If it's one of our nightly master builds (which we published recently for the bug in polygon simplification in another thread) then it should also include a date there. I wonder where you got that build from.

In any case, even in that nightly we published then, that bug wasn't fixed yet, but it is fixed for the 4.8 beta. We've just published that here: https://github.com/Ultimaker/Cura/releases/tag/4.8-beta Could you try with that version?

Gonzo3so, 2020-10-23:
I'll give it a go, thank you

Gonzo3so, 2020-10-23:
I just installed 4.8 beta and the bug is gone, perfect. 

As to why I have Cura master? I honestly don't know how or why I installed it.

fieldOfView, 2020-10-23:
Might have been an @smartavionics build. AFAIK, they are always based on the master branch.

konskarm, 2020-10-23:
> I just installed 4.8 beta and the bug is gone, perfect.
> 
> As to why I have Cura master? I honestly don't know how or why I installed it.
Excellent! We can close this issue.

closed 2020-10-23 by konskarm

Tampa, 2021-08-23:
There is still an issue with this to some degree. When rotating an object like a long stick say 35 degrees on the buildplate and then scaling with uniform scaling unchecked the result is the object actually changing shape. Image below should be a rectangle, but the corners changed angles and that seems to depend on the selected scale vectors. I think instead of scaling the object according to the original rotation, instead the new "bounding box" orientated to the buildplate itself is scaled, causing the object to actually shift along the XY axis of the buildplate and not the object. Given this is not rotated in the stl itself, but in CURA, it should probably not mess with proportions like this.

[image]

nallath, 2021-08-23:
> There is still an issue with this to some degree. When rotating an object like a long stick say 35 degrees on the buildplate and then scaling with uniform scaling unchecked the result is the object [...]

No, that is correct. The scaling is done on a world level. That's also what the tool handles indicate; They aren't rotated.

Tampa, 2021-08-23:
That creates a disconnection between the model and its rotation however as once the model is distorted from scaling at a different angle the transform sticks through further rotation or reset. Reversing the transformation means rotating the opposite direction and scaling once more, making it rather difficult and unintuitive to prepare a model for printing as to avoid problems with distortion scaling can only happen without rotation applied.

A switch to change the scaling based on the models axis rather than that of the buildplate would make this more intuitive and the original behavior can be kept as well. It's not what one expects when most cad programs and the like scale based on object origin when it comes to self contained parts.

And no just showing the handles as not rotated is not a good indication when the rotation handles don't change their indicator positions on rotation either, even if a rotation is applied their arrows still stick in the same place, which gives the impression the indicators are not meant to move to show a change in rotation or scale either, else the rotation indicators would orient according to model orientation and the scale vectors would change their position according to the applied scale factor.

Ghostkeeper, 2021-09-01:
The logic is that, from the moment you release the rotation handle after rotating your cube 45 degrees, the model behaves as if you loaded in the rotated model. There's no state. I'd say it's illogical that saving the model and re-loading it would change the way it scales afterwards.

Further confusion would be that after rotating your model 90 degrees, scaling the X dimension would result in the model scaling in the Y direction.

There is no consensus on which one to use in the industry. Some (e.g. Blender) apply transformations locally. Some (e.g. CloudCompare) apply transformations globally.