Ultimaker/Cura #8596: Not printing on Ender3 over USB Cura 4.7

DJDC13, 2020-10-22:
**Application version**
4.7

**Platform**
Windows 10 64 bit, GTX1060 6Gb, 16Gb RAM, nVME, SSD and magnetic storage, Ryzen 5 3600 and Asus Prime X570-p mb

**Printer**
Creality Ender3
**Reproduction steps**
1. Plug printer into USB and tried to print a sliced model
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
The printer does absolutely nothing other than heat the bed. The progress bar in Cura continues to go up until it reaches 100% at which point the heated bed turns off.

**Expected results**
The printer should start printing

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues 
[cura.log](https://github.com/Ultimaker/Cura/files/5424809/cura.log)

to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I have been using octopi for the last year but have moved and currently needed to use the USB connection. Added a new printer with all the same parameters as the network printer and tried to print. The printer connects, I can move the axis using the jog buttons or by typing G-code and the bed and nozzle can both be heated. But as soon as I start a print the bed heats up but nothing else happens. The nozzle also will not begin heating by itself upon starting the print.

nallath, 2020-10-23:
Well, if it's heating up, at least one of the commands reached it. 

If the progress bar is increasing, it also means that the printer is responding with an "ok" response to Cura streaming the responses to it.

DJDC13, 2020-10-23:
> Well, if it's heating up, at least one of the commands reached it.
> 
> If the progress bar is increasing, it also means that the printer is responding with an "ok" response to Cura streaming the responses to it.

Yes it is strange that I can move all the axis via the G-code and via the jog buttons in Cura but it will not start printing. If I save the sliced file as G-code and run the G-code on Pronterface it prints fine

Ghostkeeper, 2020-10-23:
As this is really printer-specific, there is not really anything we can do to help, here, without having access to that printer.

Perhaps someone else with an Ender 3 could confirm whether they have it too, or maybe it's something related to a specific firmware?

Perhaps it's something in the speed at which the commands arrive? Is it possible to try a different interface like Pronterface to see if it reacts to multiple commands in a row then?

DJDC13, 2020-10-23:
Yes as I said in the previous comment I have run the exact code on Pronterface and it prints just fine. I have tried making Cura only use the specific baud rate instead of trying each one and no difference. Previous versions of Cura have worked over USB but this is the first time I have had to use it for over a year. I have tried in a USB 2.0 port and also in 2 different gen USB 3 ports.

mschilt, 2022-02-06:
Ender 3 works great here with Cura 4.13. Does the issue persist? Can we close this?

Ghostkeeper, 2022-02-18:
The newest version didn't change anything in the USB printing functionality. The last change was in [June](https://github.com/Ultimaker/Cura/pull/8463). So if it got worse in the latest release, what changed must've been the hardware.

Ghostkeeper, 2022-02-26:
We didn't recently change anything in Cura w.r.t. the USB printing functionality, not since v4.9. If the problem got worse in the newest version of Cura, something else must also have changed. This could be firmware, hardware or the cable.

A log file could shine a light what is happening if the monitor page becomes black or causes Cura to crash.

GregValiant, 2024-11-20:
USB Printing is a legacy function that is included with Cura for older UM printers.  It isn't maintained (and hasn't been for a while).

I'll go ahead and close this.  For reference, my printer is an Ender 3 Pro and it prints fine over the USB.  I use the SD card since Microsoft pushes out updates during the night which causes the computer to reboot and makes overnight printing impossible.

closed 2024-11-20 by GregValiant