Ultimaker/Cura #8597: some problem of "Merge Models"

joonghochoe, 2020-10-23:
**Application version**
(The version of the application this issue occurs with.)
Master Version

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 V.2004

**Printer**
(Which printer was selected in Cura?)
Any Printer

**Reproduction steps**
1. load multi color model(ex. https://www.thingiverse.com/thing:4511733)
2. merge models

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
- In master version
loaded models
[image]
merged models
[image]

- 4.7.1 version
loaded models
[image]

merged models
[image]

**Actual results**
(What happens after the above steps have been followed.)
The multiple models are merged in an arranged shape.

**Expected results**
(What should happen after the above steps have been followed.)
want to be put together in the right shape.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
[CE5P_24.zip](https://github.com/Ultimaker/Cura/files/5426889/CE5P_24.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
This is the master version with the libnest2d, pynest2d  library(?) applied.

joonghochoe, 2020-10-23:
I have found a normal merge procedure.
1. load models
2. select all models
3. select rotation tool
4. reset rotation
5. merge models

nallath, 2020-10-23:
For our reference: CURA-7796

closed 2024-11-20 by joonghochoe