Ultimaker/Cura #8600: Fortis Profile Bugging Material Estimation

Mikaledoux, 2020-10-23:
**Application version**
Version 4.7

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Ultimaker S5

**Reproduction steps**
1. I imported my model in Cura.
2. Choosed my registered Fortis Profile. 
3. I Desactivated PrintCore 2 because i don't need it.
4. I sliced it as usual
 
**Screenshot(s)**

[image]

[image]

[image]

**Actual results**
In estimated material, Cura say me that PrintCore 2 is activated and that it prints almost all of my job.

**Expected results**
PrintCore2 should be turned off and my job printed on PrintCore1.

**Project file**

**Log file**

**Additional information**
This is not the first time this kind of problem has happened. But until then I simply had to reactivate the PrintCore2 and deactivate it again. It was finally working. Now I have this problem that persists.

nallath, 2020-10-26:
Could you check again with 4.8 beta? We recently fixed another issue that might influence this.

Mikaledoux, 2020-10-28:
Thank you, I'll try, and I'll keep you posted.

nallath, 2020-11-03:
Any update on this? Was the issue resolved in the beta?

closed 2020-11-17 by no-response[bot]