Ultimaker/Cura #8602: Supports missing with Gradual Support Infill Steps

AbeFM, 2020-10-23:
**Application version**
4.7.1

**Platform**
win10

**Printer**
Custom Dual Extrusion, using a single extruder

**Reproduction steps**
1) Load model (I've done this with a couple models)
2) Slice with gradual supports (50%, 2 steps but it happens with other settings)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
I noticed supports that were near the top were floating a few mm from the model, as shown. You can see the brim on the printer bed as well.

**Expected results**
Nothing should ever be printed floating in mid air. I expected a small amount of support up to a top with higher density.

**Project file**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I don't think this is an X/Y distance priority issue.
Probably adding walls - or in some cases using join supports - would be a decent work-around. I'm just reporting this because it seems like an error that could be caught. I'd rather not have the support print at all than print a truly floating support.
Join supports seems to at least connect to the rest of the support, though some printing is still in mid-air:
[image]

Also, joining supports makes a much better first layer. Much of the first layer didn't seem to reach down past support brim before.

Ghostkeeper, 2020-10-27:
I'm not really seeing a difference with Gradual Support Infill Steps set to 2 vs. 0, other than it reducing the density of support. Naturally, reducing density the lower you get will result in some support lines being suspended in mid-air. Lines and Zigzag are not very good patterns to work with gradual infill since they never cross, and thus half of the lines get suspended in mid-air at every step.

The back side of that seat is just as useless if Gradual Support Infill Steps is set to 0:
[image]
This is issue https://github.com/Ultimaker/Cura/issues/7487.

Maybe what you're talking about is that the connecting "zag" sometimes flips to the other side, leaving zags in higher layer over places where there are no zags in lower layers, like what happens a lot on the side here of your seat model:
[image]

Is that what you meant?

AbeFM, 2020-10-27:
I see what you're saying about the poor choice of patterns. I'm not convinced that's all I'm seeing. The 'gradual steps' is a guess on my part, I haven't tried with/without as you did - a smart thing to check.

[image]
I circled in black free floating supports. This is what got me thinking about the gradual steps - that you clearly are getting "brim around supports" for supports that don't touch said brim.
The very lowest highlighting circle even shows some "support floor" that has nothing above it. I suspect it's low density support causing it, gradual or not.

As you say, oftentimes, gradual support will start in mid-air, but after a couple layers it builds up, attaching loosely to nearby supports - but the islands of support I have shown look destined for failure to my eye.

Anyway, looking at it closer, I think it's a combination of the 'side switching' you're talking about coupled with the gradual infill - I can get it to happen and disappear by turning off gradual infill, but I can also get gaps where there's several free floating chunks of support in a single collum. 

[image]

It seems to me like you could/should skip anything CURA generated (not in the model) that's 100% free standing in mid-air. Might just be worth closing the thread.

Ghostkeeper, 2020-10-28:
> This is what got me thinking about the gradual steps - that you clearly are getting "brim around supports" for supports that don't touch said brim.

Indeed, it's drawing a brim around support that is too small to print (due to line width). I'd consider that a bug. The bug isn't nearly as serious as the other things you're reporting, free-floating support, but I can't seem to reproduce those, except when it's due to the infill pattern only. The support regions always seem to rest on either the model, the build plate, or on other support.