Ultimaker/Cura #8604: Support is generated in the wrong place

wolfgangmauer, 2020-10-23:
**Application version**
4.7.1

**Platform**
WIN10

**Printer**
DIY BLV mgn Cube

**Reproduction steps**
1. Turn on "Support Structure"
2. Select "Touching Buildplate"

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
The holes do not touch the build plate, but get support

**Expected results**
No support at the holes

**Project file**
[91da9961-a752-41cc-a532-eca90527c4fe.zip](https://github.com/Ultimaker/Cura/files/5431752/91da9961-a752-41cc-a532-eca90527c4fe.zip)

Ghostkeeper, 2020-10-27:
Your project file is not a project file, but an STL 3D model file. You can save a project file by going to File -> Save Project... Please submit a project file, because most likely I'm not using the same settings and profiles as you are, and perhaps oriented the model differently too.

I'm unable to reproduce this problem. Using your reproduction steps, I get no support that rests on the model:
[image]

Compare that with when it's set to "Everywhere". You get support resting on the model then:
[image]

Perhaps you misunderstood the feature. If Support Placement is set to "Touching Buildplate", support may only rest on the build plate, not on the model. It is not about whether there are any holes in the model or whether the overhang touches the build plate. Overhang that touches the build plate doesn't need support at all, since it rests on the build plate then.

wolfgangmauer, 2020-10-27:
Thanks for the explanation.

closed 2020-10-27 by wolfgangmauer