Ultimaker/Cura #8605: [4.8 beta] arrows in profile categories don't change

Liger0, 2020-10-23:
**Application version**
4.8 beta

**Platform**
Win10 x64

**Printer**
Does it even matter?

**Reproduction steps**
1. Open or close the settings category in the profile
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Nothing. They stay like they were before since... I don't know what decides their default behaviour?

**Expected results**
They animate

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Vandrasc, 2020-10-27:
Hi @Liger0 , thank you for your feedback once again :) Yes, we had this ticket in our backlog from 4.7 but we didn't have the time to get to it yet. Devs see CURA-7692 for our own internal reference.

Ghostkeeper, 2020-10-27:
Duplicate of https://github.com/Ultimaker/Cura/issues/8283.

closed 2020-10-27 by Ghostkeeper