Ultimaker/Cura #8607: [4.7.1] Crash while saving project after slicing

Garywoo, 2020-10-24:
**Application version**
4.7.1

**Platform**
Windows 10 - Version 2004 (OS Build 19041.572)

**Printer**
Lotmaxx Shark

**Reproduction steps**
1. Open any .STL file
2. Click Slice
3. Click File > Save Project
4. Click Save on the summary window
5. Choose a save location > Click Save

**Actual results**
Cura crashes. Project file is not saved.

**Expected results**
Cura doesn't crash, and file is saved

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5432338/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5432339/DxDiag.txt)

**Additional information**
If the model is not sliced, then the project will save normally without crashing.

nallath, 2020-10-26:
Does it only happen when saving a project? What happens if you save via the button in the lower right of the screen ("Save to file")?

Garywoo, 2020-10-26:
Saving the gcode file works fine, the issue is specifically with saving the project file.

Ghostkeeper, 2020-10-27:
I can slice and then save a project file fine. No crash here.

In your DxDiag I see one freeze and 8 crashes. The crashes are all segfaults in Qt5Gui.dll. Since they are segfaults in the C environment of CPython, our code doesn't get the chance to write it to the log. In the log file it just stops abruptly.

Since they are segfaults in Qt, it is certainly a bug in either PyQt or Qt itself. It will be very hard to do anything about it. We'd need to find a workaround that happens to not trigger the bug. But since we can't reproduce the crash ourselves, it will be extremely hard to iterate anything to find such a workaround.

I think the only real hope here is that we find more precise reproduction steps that allows us to reproduce the crash ourselves. Although there is a slight hope that the bug is fixed in newer versions of Qt, so updating Qt could also help.

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

Garywoo, 2024-11-21:
I've tested this again in Cura 5.8.0 and 5.9.0, and the issue is no longer occurring for me. I'll close the issue now.

closed 2024-11-21 by Garywoo