Ultimaker/Cura #8608: To add Two Trees Bluer as a selectable printer

WashingtonJunior, 2020-10-24:
**Is your feature request related to a problem? Please describe.**
TwoTrees Bluer is getting more popular each day and it's kind of frustrating when you go to add it to Cura and it doesn't show up.

**Describe the solution you'd like**
All the necessary files to do it are available here: http://gsxbr.net/updates/3d/NewCura46.rar (with instructions here: https://gsxbr.net/homunculo/?p=441)
It would be nice if they can be included in the default Cura installation, without the user having to get this file, unpack and copy it to Cura's resources folder. And yes, that's my website and I designed these files and I would allow them to be included at no charge, as I already distribute them freely.

**Describe alternatives you've considered**
The alternatives are installing it as a Custom FDM printer or as Creality Ender 3, but that makes the printer to lose its "identity".

**Affected users and/or printers**
The growing community of TwoTrees Bluer users would be thankful, as it will be easier and clearer to add it as a printer.

**Additional context**
[image]

Thanks!

konskarm, 2020-10-26:
Hi @WashingtonJunior. I see that you have already created the definitions. Please create a pull request which will include the definition files. The pull request will run our automated tests which will let us know if something is not in place with the files you want to add. In addition, it will make it easier for us to review and test your definition file before adding it to our codebase.

WashingtonJunior, 2020-10-26:
Hi @konskarm! Thanks for replying, I will do that, it's my first time doing this, but I think you explained well the process. :D

closed 2020-10-26 by WashingtonJunior

konskarm, 2020-10-26:
Happy to help! Keep in mind that it may take a while to review and accept it due to our workload.