Ultimaker/Cura #8611: Gradual Bed Temperature Control : FEATURE REQUEST

sezeryalcin, 2020-10-24:
High bed temperatures are required for most of the prints. This is for sticking well to bed and also for ambient temperature for those who use enclosure during printing.

However as you get closer to end of print, bed adhesion is already established and this temperature can be gradually reduced.
This can help with 2 things:
1) Save energy by heating bed at a lower temp
2) It will help bed cool faster once print finished. This is usually required for removing print easily on glass beds.

For this feature and to be as flexible as possible, following 2 settings can be added:
-"Gradual Bed Temperature Layer"
-"Gradual Bed Temperature Change per Layer"

"Gradual Bed Temperature Layer" :This will specify the layer number we want to start changing temperature. If it's a positive number, it will start from that layer. If it's a negative number, it will count from last layer in reverse. For example, for a 120 layer job, specifying 100 or -20  will have the same effect.
Having functionality to use negative value here will help to define this setting for most prints. Typically I would assume -5 value will be useful for most people.

"Gradual Bed Temperature Change per Layer"
This value, will be in Celsius will define temperature change. It will be usually a negative value, but allowing positive value may also make it flexible to use for other occasions. Typically I would use -1 for this setting to slowly allow bed to cool down while print is finishing.

On a typical scenario, I print PLA with 75C bed. I could easily use following values for my case, saving energy and also making print removal faster.

"Gradual Bed Temperature Layer" = -10
"Gradual Bed Temperature Change per Layer" = -3

This way, it will start reducing temperature on last 10th layer and when print finishes, bed will be 45C instead of 75C.

nallath, 2020-10-26:
Soo, like the  "Build Plate Temperature Initial Layer" setting?

sezeryalcin, 2020-10-26:
This feature will mostly be useful during final layers. Initial Layer Temperature setting is only for first layer.

Ghostkeeper, 2020-10-27:
We've seen - at least with Ultimaker's printers - that the bed temperature really only translates through about the first 1cm of the print's height. Beyond the first 5mm the ambient temperature is more dominant and after a centimetre or so the influence of the bed on the temperature of the plastic is not measurable any more.

This is probably different if you have an isolating enclosure. But then you wouldn't want to cool down the build environment during the final few layers either, or those layers would solidify differently and you'd get banding. A heated build volume is meant to improve annealing and layer bonding, and those things are not specific to bed adhesion. They still apply in the same way during the higher layers. Why would you want to drop the build volume temperature at a certain height?

Aside from the problem, there are also two problems with the solution:
* The build plate is very slow to change temperatures. A layer change can happen every 5 or 10 seconds, depending on the Minimum Layer Time, especially higher up in the print. PID regulators don't like to get new targets too often, and tend to overcompensate when this happens. This could lead to temperatures varying wildly up to 30 degrees off the target or so. We've seen this with the nozzles, which is why we have a minimum stand-by duration (default of 50 seconds).
* Having a delta temperature applied every layer is not very user-friendly as it requires the user calculating with the layer height and multiplying with the delta. I think having it gradually change to 0 over a set amount of layers is more intuitive and more in line with similar settings like Initial Layer Fan Speed.

GregValiant, 2026-09-08:
I'll go ahead and close this.  It could be done easiest in post-process and if there is enough interest I would do it.

closed 2026-09-08 by GregValiant