Ultimaker/Cura #8612: 4.8 - beta - Multi-Color in single nozzle G-Code bug

hugoalves6, 2020-10-24:
Hello, I just found a new "bug" in the beta 4.8

Every time I start a print, only in the multi-color settings, it generates this G-Code:

`;Generated with Cura_SteamEngine 4.8.0-beta
T0
M140 S65
M105
M190 S65
M104 S205
M104 T1 S150
M104 T2 S195
M105
M109 S205
M105
M109 T1 S150
M105
M109 T2 S195
M82 ;absolute extrusion mode`

For me is that my printer only have 1 nozzle, making it impossible to heat the 2nd nozzle, and it generates a bug on the printer, and it just freezes there.

Please is it possible to remove the temperatures that are auto generated?

Asterchades, 2020-10-25:
Try manually adding a command to heat the second nozzle startup script, but comment it out so the printer doesn't try to execute it. eg:

; M109 T2 S0 ;Dummy temperature line for "Nozzle 2"

Regular temperature settings can be overridden like this (such as if you want to use your printer's pre-heat option rather than setting it in the GCode) so I would assume that multi-effector settings can be similarly overridden. Worst case you may need to add commands to heat the actual nozzle as well (though obviously don't comment them out) on the off chance that adding one removes automatic creation of the other.

Ghostkeeper, 2020-10-27:
It should not generate those commands for single-extrusion printers. When I try this with the single-extruder Ultimaker Original printer, it is only generating heating commands without `T` parameter.

Could you fill in the issue template? It asks for a project file, which printer you are using and log files, all of which are useful to figure out why it is generating that g-code. Otherwise we'll close this issue.

fieldOfView, 2020-10-27:
This is about a single nozzle multiple extrusion setup, which requires you to set up multiple extruders in Cura.

@Asterchades, do you have the Shared heater option checked in the machine settings?

Asterchades, 2020-10-27:
Would be pointless of me to do so since I don't use my machine like that. I've considered it but it's not something I've been actively looking into. Was merely offering a suggestion based on how I suspect Cura functions in this specific usage case.

But whether @hugoalves6 is using that option or not is probably quite relevant.

hugoalves6, 2020-10-28:
Hello everyone, so i have 1 single extruder, with 3 colors (not mixer), I made this change on the printer, Its runing a stm32 bits board

But I don't see that shared heater option on my cura

[image]

Ghostkeeper, 2020-10-30:
The "Extruders Share Heater" setting is a machine setting that is normally not visible. It can be made visible by installing the Printer Settings plug-in, which can be found on the Marketplace.
[image]

hugoalves6, 2020-10-31:
> The "Extruders Share Heater" setting is a machine setting that is normally not visible. It can be made visible by installing the Printer Settings plug-in, which can be found on the Marketplace.
> [image]
> 
> It's intended that a printer definition would set this, if Cura has a definition for your printer.

Thx Ghost, just installed it and its working fine, appreciated it, thank you for your help 👍

Ghostkeeper, 2020-11-03:
Note that Extruders Share Heater has had some problems in the past and will need customised start and end g-codes to make it work properly: https://github.com/Ultimaker/Cura/issues/8148

hugoalves6, 2020-11-03:
> Note that Extruders Share Heater has had some problems in the past and will need customised start and end g-codes to make it work properly: #8148

Like what kind of starts @Ghostkeeper ?
im having some issue on the T changing, it overextrudes and if i compensate that overextrusion on extruder start gcode, on the first layer it under extrudes a lot (like 30mm)

Ghostkeeper, 2020-11-05:
You'll probably need to instruct to do a large retraction in the extruder end g-code of each extruder. Something like:

```
G91
G1 F1000 E-40
G90
```

And then an unretraction in the extruder start g-code of every extruder:

```
G91
G1 F800 E40
G90
```

More details are in the thread that I linked. I don't have such a printer with a mixing nozzle so I'm only theorising.
Please refer to #8148. There was also someone in that thread working on implementing that in CuraEngine so that it can work out of the box.

closed 2020-11-05 by Ghostkeeper