Ultimaker/Cura #8613: [4.8 beta] Slicing STL: Fillets sliced wrong

GrooverFromHolland, 2020-10-25:
**Cura 4.8 beta**
**Windows 10 64 bit**
**Ender 5 Plus**

When slicing STL file, some fillets are replaced with a rectangle surface in outer wall.
Version 4.8:
[image]

Version 4.6.2:
[image]

STL file:

[ShelfBracket20-24.zip](https://github.com/Ultimaker/Cura/files/5434687/ShelfBracket20-24.zip)

The two versions are using the same settings.
The STL is made with Fusion360 
Format: binary
Nr. of triangles: 53394
Refinement: High

nallath, 2020-10-26:
Could you also share a project file? I think it might be related to settings that you used.

GrooverFromHolland, 2020-10-26:
Here is the Cura 4.8 Project file:

[ShelfBracket20-24.zip](https://github.com/Ultimaker/Cura/files/5440618/ShelfBracket20-24.zip)

Ghostkeeper, 2020-10-27:
This is a problem with your settings, particularly the Hole Horizontal Expansion setting which is set to 1mm.

Hole Horizontal Expansion makes all horizontal holes wider. There is a big hole in the middle of your model, as becomes apparent when you look at a single layer:
[image]

Yes, I mean the big rounded triangle in the middle throughout 95% of the layer. That's a hole in the shape, so it gets expanded by 1mm.

A few layers lower however, the shape breaks into pieces due to the vertical holes on the side of the bracket:
[image]

Suddenly there is no more hole, just 4 rather slim parts. And thus no hole gets expanded any more. The result is a shelf of 1mm at the height where the last screw hole closes up.

closed 2020-10-27 by Ghostkeeper

nallath, 2020-10-27:
I've also tested this with 4.6; The result is the exact same. So it's also not a regression.