Ultimaker/Cura #8614: [4.8 beta] Support branches removed with z offset of 0.

Liger0, 2020-10-25:
**Application version**
4.8 beta

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Tree supports
2. Z Distance 0

**Screenshot(s)**
[image]
[image]

**Actual results**
Almost all of the tree support branches are removed

**Expected results**
Tree supports should stay there

**Project file**
[CCR10S_no z distance most supports disappear.zip](https://github.com/Ultimaker/Cura/files/5434756/CCR10S_no.z.distance.most.supports.disappear.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
This is a second issue about the usage of tree supports which makes them unusable. 
Recently, the tree supports became useless because of another issue with the tree supports, which is the tip of them is horizontally too far from the piece they support, thus not supporting anything.
Nothing helped with this: Reducing the minimum are for support, using an x-y distance from the model of 0.1.
[image]

Asterchades, 2020-10-26:
> Recently, the tree supports became useless because of another issue with the tree supports, which is the tip of them is horizontally too far from the piece they support, thus not supporting anything.

Disable the Support Interface - or at least the Support Roof. It's currently incompatible with Tree Supports, causing exactly that problem.

Liger0, 2020-10-26:
> Disable the Support Interface - or at least the Support Roof. It's currently incompatible with Tree Supports, causing exactly that problem.

The problem is that since cura 4.7 the normal support share the same settings. 
I would like to make the tree supports not having the interface layer and not starting from the model (the opposite of what I have on normal supports), but instead this must be selected each time. Pretty weird since the infill can be selected separately.

But yes, as you said that helps a lot horizontally wise.

Asterchades, 2020-10-26:
To be completely fair, the Interface settings were shared prior to 4.7, too. The only real difference (to the end user) beyond where the two support settings are located is that it is no longer possible to activate both types of supports at the same time.

Unfortunately, while 4.7 made Tree Supports very consistent, for much the same reasons you've noticed I don't personally agree that it did enough to justify removing them from Experimental. But I guess consistency is key to start gathering relevant feedback like what you've provided, and features buried in Experimental probably don't get the attention they should from general users.

Ghostkeeper, 2020-10-27:
We did get a lot more feedback with tree support since 4.7. In great part from the people involved in this topic :smile: And a lot about smaller details with tiny parts resting on vertical slopes and such. The move out of experimental was meant more as a tech push ("let's make tree support stable so that people can use it to save printing time and material") rather than to get more feedback.

I've never really seen the point of using support interface with tree supports. Tree supports already generate sort of an interface by converting the branches to slim lines, which is less configurable but does fit the shape of tree support better. As a user you can create profiles that combine tree support with disabling support interface, which is a bit cumbersome. A profile designer can actually make the interface disable automatically if the user selects tree support though. If you guys, as users of Creality CR10S, think that this works better for your printers, I can totally make that happen.

The bug itself is indeed correct. I don't know what exactly the cause is here but it seems to have started with the fix for tree support trying to rest on vertical slopes after being misclassified as not being able to reach the build plate.

Liger0, 2020-10-27:
To me tree supports are very important to use in conjunction with soluble interface layers, otherwise I can't make any use of them.

Ghostkeeper, 2020-10-29:
There is a similar case in https://github.com/Ultimaker/Cura/issues/8644. It seems that the support branch is immediately removed if it is created within the X/Y distance of a model's outline. If the Z distance is low or the wall steep (=not extreme overhang) then the branch is created horizontally very close to the model, and then consequently removed.

MariMakes, 2023-07-31:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue with the tree support branches.

In Cura 5.4 we introduced a new tree support I tested this issue, and it seems that it's resolved. 
[image]

You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍 

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
I'll be closing this issue. I hope you understand.

closed 2023-07-31 by MariMakes