Ultimaker/Cura #8615: Wrong comment in generated gcode regarding brim or skirt settings

GeVanCo, 2020-10-25:
I recently stepped into the world of 3D printing and bought me a Balco Touchscreen 3D printer (which is in fact a Wanhao Duplicator i3 Plus version sold by Balco).  The slicer software which is part of the packet is Ultima Cura 3.1.4.

When starting the slicer and ask for an skirt or brim, the generated gcode always shows "Skirt" in the comments.  However, the 3D print itself is OK and is starting with a skirt or brim, whatever is chosen.

Then I downloaded the very latest Cura 4.7.1 to see if that issue is still there and yes, it is.  Whatever you choose, skirt or brim, the resulting comment in the generated gcode is "Skirt".  But again, the 3D print is correct.
This is, however, confusing at first.  Certainly for a NOOB like me in the realm of 3D.

This isn't a big issue but it would be nice if the gcode would contain what is selected.

Best,

--Geert

Ghostkeeper, 2020-10-27:
Indeed, the comments are not very granular. The brim and skirt are to Cura really the same, just different settings. There will also never be a brim and skirt at the same time (except if Support Brim or Prime Tower Brim is enabled and the adhesion is set to "skirt"). Consumers of g-code don't generally need to distinguish the two.

It would've been better to change this to say "adhesion". But changing that would break other programs that depend on these comments, such as OctoPrint or some post-processing scripts out there. I don't think we can really change that any more now unless there is a serious problem with it.

Ghostkeeper, 2020-10-27:
Thanks for the feedback though! You are certainly not a noob in 3D if you're already looking into the g-code. And I bet you've actually been 3D for years!

closed 2020-10-27 by Ghostkeeper