Ultimaker/Cura #8617: [4.8 Beta] App does not start on MacOS BigSur

Renha, 2020-10-25:
**Application version**
4.8 Beta

**Platform**
MacOS Big Sur

**Printer**
None

**Reproduction steps**
1. Install Cura
2. Launch Cura

**Actual results**
Cura jumps couple of times then nothing happens

**Expected results**
Cura launches just as 4.7.1

**Log file**
No log file for version 4.8 present

evtrados, 2020-10-26:
Hi @Renha,

Thanks for reporting this. Unfortunately none of us has Big Sur installed as yet (we still use Catalina), so I would like some more information:
- Did the OS give any warnings while opening the .dmg or while verifying Cura?
- Can you please do the following:  
  1. Open a terminal window and navigate to where the Ultimaker Cura.app is. That should be: `cd /Volumes/Ultimaker_Cura-4.8.0-beta-Darwin` 
  2. Then enter `open Ultimaker\ Cura.app`

Since the icon jumps a couple of times as you said, I suspect that either Big Sur does not/cannot verify Cura as a safe application or there's something else obstructing it from launching.
When I do the above steps this is the Window I get, notifying me that the app is safe and nothing malicious was found:
[image]

marcelonovaes, 2020-10-27:
I can confirm that it doesn't shows a similar dialog, and the app opens and closes.
I've noticed similar behavior with other applications using QT. there's a version of QT that relies on GL which I believe BigSur is Metal only.

evtrados, 2020-10-27:
I was very curious, so I installed Big Sur on an old MacBook I had lying around.
I do get the same dialog though (see below), but after the verification is successful the app indeed does not start. There's no error thrown whatsoever, so we're going to discuss this with the team and see how we can approach it. I don't think it is related to QT, since Cura 4.7.1 launches without issues.
[image]

fieldOfView, 2020-10-27:
Surely something must be logged in Console.app (the system-wide log, not the Cura log)

suryanisaac, 2020-10-27:
Hey, it works with the geos.pyc file provided for the 4.6.2 Big Sur fix (https://github.com/Ultimaker/Cura/issues/7970#issuecomment-659791641). Hopefully this helps.

I changed my geos.pyc file to the provided one and Cura launches fine for me.

Ghostkeeper, 2020-10-27:
Hmm, we did apply that patch though I think. Maybe this was not applied correctly but somehow manually or something?

suryanisaac, 2020-10-27:
Maybe, I need to take a look at the source code. Do you have any idea where this file is in the source code?

evtrados, 2020-10-27:
@fieldOfView , yes Console was showing an error on a `/Users/$USER/Library/Logs/cura/stderr.log` file from 24 Oct, so I didn't bother looking into it, because that was 2 days before I upgraded. Once I navigated to its path though I saw it had changed like 1 min ago. 
And, indeed, the problem is that geos cannot be found:
`OSError: Could not find lib c or load any of its variants [].`

marcelonovaes, 2020-10-28:
I can confirm that copying geos.py from 4.7.1 to 4.8 Beta works on Big Sur

suryanisaac, 2020-10-28:
Thanks! I never downloaded 4.7 because of the bug with curves. Good to see that it works.

jeff-bravo, 2020-10-28:
This fixed the issue for me as well. Adding [geos.py](https://github.com/Ultimaker/Cura/issues/7970#issuecomment-659791641) to `Ultimaker Cura.app/Contents/Resources/lib/python3.5/shapely/` restored normal app launch under Big Sur.

evtrados, 2020-10-28:
We have also applied the patch on our build system. Cura v4.8 stable will have the fix and will, thus, launch without issues on Big Sur. Thank you all!

suryanisaac, 2020-10-29:
Thanks for your help!

truegret, 2020-10-29:
Thank you!

How about Yosemite?

David

iPad iOS 14.1 (18A8395)

Ghostkeeper, 2020-10-30:
We weren't aware that there was a problem when running on MacOS 10.10 Yosemite?
What are your symptoms, Truegret? Do you have any logs or crash log?

evtrados, 2020-11-03:
As far as I recall, Yosemite is the oldest version we currently support, so all should be working fine there. The furthest version I can go back to is El Capitan 10.11, where I never encountered any issues with the .dmg or the .app.

truegret, 2020-11-13:
Please consider re-opening this issue.

Unless you expect me to wait until Ultimaker-Cura 5.0.0.

suryanisaac, 2020-11-14:
Yeah, this needs to be opened

Ghostkeeper, 2020-11-17:
Does this issue also happen with the 4.8 stable? Or just with the beta?

Renha, 2020-11-19:
I will be able to check that later today, probably someone will be faster

Renha, 2020-11-19:
The issue is solved in stable version of Cura 4.8, thank you

closed 2020-11-19 by Renha

PlasmaSoftUK, 2020-11-21:
Still not working for me on Big Sur, and the current website download

Ghostkeeper, 2020-11-23:
Do you get the same symptom? Anything in the logs?

PlasmaSoftUK, 2020-11-24:
> Do you get the same symptom? Anything in the logs?

It doesn't create a log, it hasn't even created a '4.8' directory in /Users/plasma/Library/Application Support/cura/

Icon just bounces and never launches, I also tried removing all the older cura directories from the above link. Still does nothing.

Plasma

fieldOfView, 2020-11-24:
There are more logs than Cura logs. Something must be logged in Console.app (the system-wide log, not the Cura log)

PlasmaSoftUK, 2020-11-24:
Never mind I am an idiot! :( Looks like at some point I renamed "Ultimaker Cura" to just "Cura" so after installing 4.8 my shortcut was still trying to launch 4.6, which I realised after applying the geos.py fix above. I am now working on 4.8, apologies for the mistake.