Ultimaker/Cura #8618: Cura 4.7.1 fan speed is always at 100% in gcode

coding-pivo, 2020-10-26:
**Application version**
4.7.1

**Platform**
Windows 10 64Bit

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. Setting fan speed in the cooling option has no effect. In the gcode there is always the command M106 S255 instead of e.g. M106 S191 for 70% fan speed.

nallath, 2020-10-26:
Have you had a look at the "Maximum Fan Speed" setting? It could be that the minimum layer time is being hit. 

I've just tried to reproduce your issue with a model that does not trigger the min layer time, and I couldn't reproduce it (I get the correct M106 values set in the g-code)

Ghostkeeper, 2020-10-27:
It would be useful to see a project file that reproduces this problem.

coding-pivo, 2020-10-29:
Sorry for confusion. I figured out the correct setting in Cura and it was my fault. I need to set the starting speed explicitly to avoid speed fan at 100% on first layer and going down to nominal speed at third layer firstly. So setting starting speed (for first layer) and nominal speed to 50% will result in the correct values in the gcode.

Ghostkeeper, 2020-11-02:
Yup, Initial Fan Speed is a separate setting.

closed 2020-11-02 by Ghostkeeper