Ultimaker/Cura #8619: > let people make up their own profiles if they want to,

FuraoEvil, 2020-10-26:
Dear,

I think there might be a misunderstanding, I do have the Ender 5 Pro and I see the eSun PLA+ material. The thing is that even that I can do the settings manually and print, I can't create personal presets(profiles) and save them.
In the quality field to the right the Profile pulldown says "Unsupported", I can still pull it open and click on "Manage Profiles" it opens the Preferences > Profiles menu.
But there is no button to create even a custom profile.
Choosing a PLA material I can do duplicate of one of the default Profiles and that way create my personal Profile indeed.
So yes, as long as the PLQ+ has no default profiles impossible to create a custom one, at least in the GUI or am I missing something.

> let people make up their own profiles if they want to,

I am going to try this one last time. That is exactly why it's there; So people can make them & experiment. We hope that if people find these settings that they share them with us (and the rest of the world)

> No, no I can't because there's NOTHING, I have to start from scratch. just a blank "unsupported" with no settings for various profiles.

There is something. A material can also have settings even if there is no quality for it. You indeed have to start from scratch. But would it be better if we would prevent you altogether from doing this? Or should we break all the setups for people who have it working now? I don't think that that is a good idea.

Anyhow, we are repeating the same statements over and over again, so unless there are any new arguments about it, let's just agree to disagree.

_Originally posted by @nallath in https://github.com/Ultimaker/Cura/issues/7276#issuecomment-708215393_

nallath, 2020-10-26:
I just tried this in 4.8 and it allowed me to create profiles just fine:

[image]

Is this not possible for you?

FuraoEvil, 2020-10-26:
I am still on 4.7.1
I will check in the 4.8
[image]

FuraoEvil, 2020-10-26:
In 4.8.0 beta still the same 
"Create profile from current settings/overrides..." is still greyed out

FuraoEvil, 2020-10-26:
Trying to import a personal profile I exported on a supported material shows this
[image]

nallath, 2020-10-26:
You will first need to change a setting. Once you did that, the option will be enabled.

FuraoEvil, 2020-10-26:
Well that did the trick :)
Maybe the fellow from the other thread just needs this little detail as well ;)
Thanks for helping, I am going to try some finetuning
How could I contribute my findings for PLA+ to the project?

Ghostkeeper, 2020-10-27:
> How could I contribute my findings for PLA+ to the project?

For that you'd need to create quality profiles. Quality profiles are slightly different from the custom profiles you create through the interface, but the file format is much the same.
For an example of a quality profile, here is the profile for ABS for Creality-brand printers, 0.4mm nozzle: https://github.com/Ultimaker/Cura/blob/master/resources/quality/creality/base/base_0.4_ABS_standard.inst.cfg . If you look in your configuration folder (Help -> Show Configuration Folder...) you'll find there a folder `quality_changes` containing your custom profiles. You can open those files with a text editor. Figuring out the metadata should be semi-obvious, but keep in mind that references to other profiles are by file name (not the user-visible name) and also the setting keys are internal names as can be found in [fdmprinter.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/quality/creality/base/base_0.4_ABS_standard.inst.cfg) . You can make profiles for PLA+ by setting `material = <any PLA+ material>`.

closed 2020-10-27 by Ghostkeeper

Sebazzz, 2020-10-29:
I think that in the case of PLA+ one could argue that it is PLA, so either it needs to be marked internally as PLA or for printers we could submit pull request for PLA+ quality profiles?

Ghostkeeper, 2020-11-02:
Indeed, either of those.

There are already specific PLA+ profiles for some printers though. So changing the material type to PLA will destroy those profiles then, which is not something I think you should do unless you can show that the PLA profiles work better for PLA+ than the PLA+ profiles for all of those printers.

So then just one option remains, really.

Sebazzz, 2020-11-02:
There is a third option but that would require code changes: Allowing for PLA+ to fall back to PLA if no PLA+ profile is present.

Is there a better option than just copy/paste the PLA profile as PLA+?

Ghostkeeper, 2020-11-04:
Implementing a fallback like that would need to make changes to the file format of the material XML files to store the fallbacks, e.g. to turn the material type metadata field into a list. This is both difficult to bring about (requiring changes in the firmware of printers to properly parse) and illogical (why would a material be both ABS-type and PLA-type at the same time?). It can be done, but I don't think it will be done.

A better option than to copy profiles is to merge the common settings to a higher tier of profiles. If both PLA and PLA+ print well on your printer with, for instance, Alternate Extra Wall enabled, then it's better to enable that setting in the profile for the nozzle. If all available nozzle profiles are better off with that setting enabled, then it's better to enable that in the printer definition. If it turns out that indeed PLA and PLA+ need the same profiles and those are the only profiles enabled, I think it makes sense to put as many settings as possible in the printer definition and let the profiles just be almost an empty shell of metadata.