Ultimaker/Cura #8621: Multiply Selected Model: Allow setting minimum distance and add automatic maximum count

jt-github, 2020-10-26:
In the _Edit_ menu, the _Multiply Selected Model_ function creates and automatically positions many new instances of an object.

Add an option to the popup dialog and/or settings that would allow a user to choose the **Minimum Distance** between objects.  They are currently placed a safe distance apart, but it would be convenient to place them closer together to maximize the number of parts that can be printed.

In the same popup dialog, add an **As Many as Possible** option below the **Number of Copies** option (with a radio button in front of each to allow the user to select one or the other option).  Selecting **As Many as Possible** would automatically calculate (and ultimately generate) the maximum number of copies that could fit in the available empty bed space.

For now, it is necessary to just guess on how many copies can fit and then manually drag around the copies to squeeze them in tighter... a cumbersome and time consuming process.

### Dialog Mockup

- Number of Copies [_____]
- As Many as Possible

Minimum Distance [_____]mm

### Settings Mockup:

Default minimum distance between objects when multiplying models: [_____]mm

This request is related to the following issue:
[#7284: Multiply object as a grid](https://github.com/Ultimaker/Cura/issues/7284)

konskarm, 2020-10-26:
The "Fit as many as possible" feature when multiplying is already in our backlog (devs see CURA-7587). I'll put up the "Setting minimum distance" for discussion with the team.

konskarm, 2020-10-27:
@jt-github We decided to not implement the option for changing the minimum distance, as the new arrange algorithm has it set at 1mm, which we think is already small enough. Nevertheless, the "Fit as many as possible" is something that we will add at some point.

printingotb, 2020-10-29:
May I suggest an option to be able to keep the objects orientation? 

The first model that I drag into Cura is imported in the correct orientation. But every time that I add or multiply the model, some of the new objects are created at a 180° angle (front becomes back).

This messes up the layer seam placement in my current project as it needs to be on a specific edge of the model. So now I have to rotate and place all objects manually anyway.
Using the arrange all models function rotates some of them again.

Ghostkeeper, 2020-10-30:
You may, but preferably in a separate thread. Nobody is going to be able to find your feature request here :stuck_out_tongue:

Username1414, 2023-05-13:
I also wished there would be a setting for deciding the distance between model-copies. Not to make it any smaller (this is perfectly fine as it is), but to make it larger for some prints.